hathach/tinyusb #2: USB3.0 (xHCI) driver support

31kothari, 2017-03-10:
Hi,

May I know by what time, xHCI driver will be available in tinyUSB?

Thanks for your code and looking for xHCI driver support for Bare Metal.

Thanks!

closed 2018-03-31 by hathach

kaushalmp, 2022-09-23:
Any updated in this request ?
is there a plan to support xhci in nearest future ?

zhugengyu, 2022-12-26:
Hello, we may try this https://github.com/sakumisu/CherryUSB/tree/master/port/xhci first, then consider merge to tinyUSB,  since it's all for MCU/SoC  @hathach @31kothari @kaushalmp

zhugengyu, 2022-12-26:
I have try xhci on CherryUSB for a while, function like mouse/keyboard/mass-storage are okay, but have some touble handling hub, I have little experience on Hub, if some one handles help ...