hathach/tinyusb #9: Is gcc or clang a supported toolchain?

mithro, 2018-11-14:
The README lists the following supported toolchains;
- lpcxpresso
- Keil MDK
- IAR Workbench

I'm guessing that GCC probably already works but isn't included in the list? What about clang?

hathach, 2018-11-14:
ah, yeah, I am having major changes on develop branch. I am too tired to keep up with these IDE changes and configure. Eventually, there will be makefile (gcc, not sure about clang) and segger embedded studio for debugging. SES is crossplatform and free for non-commercial without code limit, and support lots of mcu. GUI is decent but debugging is very good. There is still lots of works though :(

hathach, 2018-12-12:
thanks @tannewt  for the anwser

closed 2018-12-12 by hathach