hathach/tinyusb #29: DFU class

hathach, 2018-12-26:
should be used with dfu-util

gimbas, 2020-06-16:
What is the ETA for this? is anyone planning on picking this up?

I wanted to make a dfu bootloader, should i just implement it myself in said bootloader?

chegewara, 2020-06-16:
https://github.com/hathach/tinyusb/tree/master/examples/device/dfu_rt

gimbas, 2020-06-16:
I get that it is partly implemented already, but the most important parts are missing, or am i equivocated?

hathach, 2020-06-17:
Please don't ask about ETA, there is none on any features/issues. People (or at least me) implement one whenever they think it is appropriate/needed and having enough time to do so.

gimbas, 2020-06-18:
I didn't want anyone to tell me when it would be done, I souldn't have said ETA, I just wanted to know if anyone was working on it, or planning on picking it up.

Thank you anyway.

hathach, 2021-07-22:
implemented since #950

closed 2021-07-22 by hathach