hathach/tinyusb #33: gcc version 8 compiler warning in msc_device.c

dhalbert, 2019-01-23:
When building circuitpython *with gcc 8.2.1* (arm toolchain 8-2018-q4), I get this fatal warning:
```
../../lib/tinyusb/src/class/msc/msc_device.c: In function 'proc_builtin_scsi':
../../lib/tinyusb/src/class/msc/msc_device.c:273:7: error: 'strncpy' output truncated before terminating nul copying 8 bytes from a string of the same length [-Werror=stringop-truncation]
       strncpy((char*) inquiry_rsp.vendor_id  , CFG_TUD_MSC_VENDOR     , sizeof(inquiry_rsp.vendor_id));
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../lib/tinyusb/src/class/msc/msc_device.c:274:7: error: 'strncpy' output truncated before terminating nul copying 16 bytes from a string of the same length [-Werror=stringop-truncation]
       strncpy((char*) inquiry_rsp.product_id , CFG_TUD_MSC_PRODUCT    , sizeof(inquiry_rsp.product_id));
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```
I tried a few ways to fix it, first using `memcpy()` and it seems like `CFG_TUD_MSC_PRODUCT` is not the same length as `inquiry_rsp.product_id`, which causes various issues.

Only gcc version 8 is catching this issue. We haven't moved up to gcc 8 yet, but we will sooner or later.

hathach, 2019-01-23:
thanks Dan, somehow I couldn't install gcc v8 via apt package. I will fix this later on.

dhalbert, 2019-01-23:
The ppa is supposed to have the latest gcc8, but it doesn't. I opened a launchpad question about it, and it hasn't been answered. I tested with gcc8 by downloading the tarball from [here](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads) and just pointing my PATH to its bin directory.

hathach, 2019-01-23:
Maybe, they didn't update some meta data. I will wait a few days, just being lazy to do reconfiguring now 😊😊

nickzoic, 2019-03-02:
G'day!  I bumped into this too (see reference above) ... looks like you're doing the right thing in copying exactly 8 bytes but GCC 8 is being a bit more paranoid.  Perhaps what we should be doing is memcpy *up to* 8 bytes from the vendor CFG string and padding any left-overs with ASCII SPACE as per this seagate document:

From https://www.seagate.com/files/staticfiles/support/docs/manual/Interface%20manuals/100293068k.pdf page 99:
>
> T10 VENDOR IDENTIFICATION field
> The T10 VENDOR IDENTIFICATION field contains eight bytes of left-aligned ASCII data (see 5.4.2) identifying the vendor of the product. The T10vendor identification shall be one assigned by INCITS. Bytes 8 through 15 contain the ASCII characters that represent “SEAGATE” (53h 45h 41h47h 41h 54h 45h 20h (space)).
>
> PRODUCT IDENTIFICATION
> The PRODUCT IDENTIFICATION field contains sixteen bytes of left-aligned ASCII data (see 5.4.2) defined by Seagate. Bytes 16 through 31 indicate the drive model with 20h (space) used as a filler. The table below is an example of drive test data returned by the drive. Bytes 16 and 17 will contain 53 54 for all drive models.

gpshead, 2019-03-04:
For circuitpython's purposes you _could_ just replace the initialization code above these errant `strncpy` usages with direct assignments to CFG_TUD_MSC_VENDOR and PRODUCT as they're #define'd as string literals and get rid of the strncpy for those two.

that doesn't address the real issue though... It sounds like the code should initialize the struct with space characters in these three fields and use memcpy to copy from the CFG_ macro values.

I made a PR that I believe does the right thing based on @nickzoic 's comment.

commit f23affbb (2019-03-04): Merge pull request #39 from gpshead/strncpy-msc_dev_ids

Fixes #33: Remove invalid use of strncpy()
commit fa8141f3 (2019-03-04): Fixes #33: Remove invalid use of strncpy().

This was causing a stringop-truncation compiler warning in gcc 8 when
the #defined values being copied from were string literals.
 `error: 'strncpy' output truncated before terminating nul copying 8 bytes from a string of the same length [-Werror=stringop-truncation]`

These fields aren't NUL terminated C strings, they are a fixed width buffer
that is supposed to be space (0x20) padded.
closed 2019-03-04 by hathach