hathach/tinyusb #35: (request)

MiliasColds, 2019-02-17:
finish custom device support

I'm working on a project that needs about 1.2 mb/s of bandwidth over USB in circuitpython which uses this USB layer. I am not very experienced with all the ins and outs of USB. I am fairly certain I can implement the bindings between circuit python and tinyusb.

As such it would be very nice if the custom_device support was finished

hathach, 2019-02-20:
Sorry for late response, have been busy with nrf52 arduino code. The custom endpoints are easy, give me a bit of time, I will add them. Note: you need to update descriptor to include those. Another option is using control transfer ( custom/vendor packet). Anyway, I will do it asap

MiliasColds, 2019-02-20:
Thanks. I thought the custom was the same as vendor specific from reading through the specifications of usb.

hathach, 2019-02-20:
Yeah, it is the same, I mean you could either have a pair of in/out endpoints as custom class or use the custom control transfer which require no more endpoints.

hathach, 2019-07-26:
Vendor class is being added (almost complete) to support webusb example #86

closed 2019-07-30 by hathach

hathach, 2019-07-30:
@MiliasColds there is more refactor to do, but it is working

MiliasColds, 2019-07-31:
Thank you for your response.

I look forward to trying it out

commit 1e7f1de4 (2019-07-30): Merge pull request #89 from hathach/develop

resolve #35 (vendor class), implement #86 (webusb)