hathach/tinyusb #41: Would love a Mynewt sample app

sterlinghughes, 2019-03-17:
It would be great to have a sample app/target that included tinyusb and mynewt on a NRF52840 (or STM32?)

hathach, 2019-03-17:
Hi @sterlinghughes, thank you for creating this issue. I did spend some time to get it work with mynewt OS, even create an repo for mynewt example here 
https://github.com/hathach/mynewt-tinyusb-exmaple

though I get a bit lost with configure and yaml, I should do better with reading docs. Then keep being pulled into other project. I will definitely give it a try later on :)

hathach, 2019-03-21:
Hi @sterlinghughes, I just updated  
https://github.com/hathach/mynewt-tinyusb-exmaple
with pca10056, the app could enumerate as both CDC and MSC device, though it requires to modify 2 places in the mynewt-core to include `nrfx_power` module for detecting USB VBUS state changes. It is probably not the right way to do so, but it runs as proof of concept and is a good start for now.

Another issue is pca10056 keep reseting every 30 seconds or so, possibly wdt or something. I will call it a day for now, and switch to troubleshooting my other issue a bit before going back to mynewt example. Please let me know what do you think and what should I change to get it better integrated with mynewt.

PS: I couldn't find an same size queue message send/receive with mynewt, thus writing the osal a bit over-complicated. If you have time, please give me your input on osal port here
https://github.com/hathach/tinyusb/blob/master/src/osal/osal_mynewt.h#L119 

for reference osal_freertos is here https://github.com/hathach/tinyusb/blob/master/src/osal/osal_freertos.h#L113

Thank you very much

hathach, 2019-07-09:
https://github.com/hathach/mynewt-tinyusb-example is updated to compile without modifying mynewt-core. Hopefully it could help you getting started with the stack. Let's me know if you need any other features/examples.

closed 2019-07-09 by hathach