hathach/tinyusb #47: tud_cdc_connected true after disconnect

dhalbert, 2019-03-22:
This is on atmel-samd. As described in https://github.com/adafruit/circuitpython/issues/1681, a CDC-writing routing in CircuitPython (https://github.com/adafruit/circuitpython/blob/master/supervisor/shared/serial.c#L54) calls `tud_cdc_connected()` to check whether it's OK to write to CDC. That routine returns `true` if CDC was connected at one time but then disconnected. I used a USB data/power-only cable to check. The `tud` CDC write routines never write anything, and so this routine never exits.

Can `tud_cdc_connected()` test more accurately for a connection? Maybe the only way to check is to try to do a write? If that's true, the write routine could set a "not connected" flag that overrides the DTR check that `tud_cdc_n_connected()` does.

hathach, 2019-03-25:
@dhalbert it should already accurate as possible, the connection state is dictate by host both connected or disconnected. Maybe the disconnection happens in the middle of writing loop here
https://github.com/adafruit/circuitpython/blob/master/supervisor/shared/serial.c#L63

Could you move the tud_cdc_connect() check into the loop() to see if that could improve

dhalbert, 2019-03-25:
I believe I called `tud_cdc_connected()` by hand from gdb and it was still true even when disconnected. I will check this. My jlink is currently tied up, but maybe I can use the pca10056 on-board jlink.

hathach, 2019-03-25:
@dhalbert when host disconnet/connect. It will send an control packet, dcd will register that event. But tinyusb only process it in tud_task i.e usb_background. If you are stepping with gdb withouth invoking the background, it may not be correct. Though there may be other bug with samd port as well, doing a full review with it just now :(

dhalbert, 2019-03-25:
In this case, I just switched the data lines off, so the host could not send anything, basically simulating an unplug. So would the control packet flow from the on-chip USB peripheral that there was a disconnect?

dhalbert, 2019-03-25:
I do agree any loops that would otherwise hang should re-check connected status if they are not making progress.

hathach, 2019-03-25:
> In this case, I just switched the data lines off, so the host could not send anything, basically simulating an unplug. So would the control packet flow from the on-chip USB peripheral that there was a disconnect?

It should, if the usb_background() is invoked to handle that usb unplug, which in turn will reset the whole cdc class driver (along with all other class drivers as well).

hathach, 2019-03-25:
hmm, there is still chance for something happened between event on the bus and background task invoked. Maybe we should put some event into priority list, which is handled insided the isr context. :thinking:

dhalbert, 2019-04-04:
I tried PR #51 on a PyPortal (SAMD51), connected via a switchable power-only/data cable. I'm running this test program:
```
import time
import supervisor

for i in range(10000):
    print(i, i, i, i, supervisor.runtime.serial_bytes_available, supervisor.runtime.serial_connected)
    time.sleep(1)
```
`supervisor.runtime.serial_bytes_available` calls `tud_cdc_available()`.
`supervisor.runtime.serial_connected` called `tud_cdc_conncted()`.

Because this is a board with an attached display, I can see the `print()` output on the built-in screen, even if CDC is not connected.

I have the following scenario:

1. Switch the cable to data.
2. Start the program, without a CDC connection. `supervisor.runtime.serial_connected` is `False`.
3. Make a CDC connection (`picocom /dev/ttyACM0`). `.serial_connected` starts printing as `True`.
4. Switch the cable to power-only. `picocom` exits. But `.serial_connected` still prints as `True`.
5. Switch the cable back to data. `.serial_connected` notices and starts printing as `False`.
6. I can reconnect via `picocom`; then `.serial_connected` starts being `True` again.

So it appears tinyusb is not detecting the physical disconnect, but does notice when I reconnect and there is no host CDC connection via `picocom`.

dhalbert, 2019-04-04:
Here is a draft PR showing what I'm running. It's very similar to yours plus checking for connected all during a write: https://github.com/adafruit/circuitpython/pull/1748

hathach, 2019-04-05:
> 4. Switch the cable to power-only. picocom exits. But .serial_connected still prints as True.

Hi Dan, thank you very much for testing this out. This should be fixed (but obviously not), maybe I missed something. Can you change the code slightly so that It will set the simple red LED (PA16)   according to the connection state. I will try to do the test here as well.

dhalbert, 2019-04-05:
OK, @hathach and I tested this, and the code is working fine. The data/charge-only cable shorts the D+/D- lines when the switch is changed, which is how a charge port is specified for USB. So that's a weird case. If we power a board separately with a battery or separate supply, unplugging the USB plug will change `.serial_connected` as appropriate. So false alarm.

dhalbert, 2019-04-05:
Fixed by #51.

closed 2019-04-05 by dhalbert

203Null, 2022-06-10:
I observed that tub_ready() returns true after tud_disconnect() on ESP32S3

Is this intended behavior?