hathach/tinyusb #52: Documentation vague about when to call tud_task

bluelightning32, 2019-04-07:
The documentation says "you need to continuously and/or periodically call tud_task()/tuh_task() function." Exactly how often does it need to be called?

I'm trying to build a keyboard using an nrf52840 that is sometimes a USB keyboard and sometimes a BLE keyboard. A low idle power usage is very important. Looking at the code, it appears that tud_task is only reading items from a queue and processing them. I don't see any timeout processing in `tud_task`. So the only way events could end up in the queue is from interrupt handlers.

Is it safe for my application to sleep indefinitely using `__WFE`, then call `tud_task` once after every wake up (due to an interrupt)? Could I just run `tud_task` from the interrupt handlers?

hathach, 2019-04-08:
No, you shouldn't and it is not safe to call tud_task in IRQ. You can put it in the main() loop and call `__WFE` at the end of the loop. USB event will wake up mcu, push event to the queue. And the task will process it and then go to sleep afterward with next wfe.

The tud_task() need to be run to handle event from usb bus such as bus event, transfer complete etc ... otherwise device won't response to host request for data. But there is no harm if you call it later other than latency.

hathach, 2019-04-15:
@bluelightning32 if it is still not clear, feel free to re-open or comment for further discussion.

closed 2019-04-15 by hathach

rlcamp, 2025-05-29:
It appears that on RP2350, tud_task() is effectively a SEV due to use of the pico spinlock in the critical section, which inhibits the WFE and turns the main loop into a busy loop, consuming a lot of power.