hathach/tinyusb #55: nRF status stage xfer complete dropped

ab932, 2019-04-18:
https://github.com/hathach/tinyusb/blob/074898099da6bfd8784019d97c64712f73b779b0/src/portable/nordic/nrf5x/dcd_nrf5x.c#L281

Doesn't this not do anything since the len is 0? This is dropped if it's on EP0.

hathach, 2019-04-18:
it does zero length transaction, most likely for control status stage

ab932, 2019-04-18:
my point is the event is dropped:

https://github.com/hathach/tinyusb/blob/0848c462b3e431a9da42e96537d2b597a4579636/src/device/usbd.c#L664

dcd_event_handler skips processing it. so you never get to execute the xfer

i'm trying to port a chip, but i currently do not know if i get an interrupt for status completion or not. unfortunately the manual for this chip is in Chinese so I'm kinda on my own here :)

i have a usb analyzer so I'm using that to help

hathach, 2019-04-18:
Not really, the DCD_EVENT_XFER_COMPLETE is called when the transferred is complete. For control status stage (zero length data), the stack has nothing to process in upper layer. So it simply skips that specific event. 

To trace/debug your port, you should have an debugger such as jlink, or increase some variable counter, and printf it later etc... Though It is much easier to use jlink along with segger embedded studio which is mainly what I use for debugging :D

ab932, 2019-04-18:
Unfortunately this chip is not an ARM chip, it is NDS32! I asked the guys using Google translate but they said that the debug port is not open... damn. Anyway, I'm slowly getting the idea that the interrupts really mean the transfer is done already (SETUP/IN/OUT). I believe I need to control the responses... Thanks for the help. I'll close this issue.

closed 2019-04-18 by ab932