hathach/tinyusb #58: Bi-directional Raw HID?

PTS93, 2019-04-25:
I was wondering if bi-directional HID is already working and if you tested or have an example for raw HID?
It's a feature I'm desperately missing and the current state of the Arduino USB stack is really not helpful in implementing it, so I was hoping tinyusb might be finally the fix for the sad state of USB on Arduino :)

hathach, 2019-04-26:
@PTS93 currently not yet. Output report can be done via control endpoint (e.g keyboard LED) since it is rarely used. Adding output endpoint for HID is easy enough, I just need a reason/scenario to do that.

For Arduino library, I am currently working on it here https://github.com/adafruit/Adafruit_TinyUSB_Arduino still require a bit more works, it is currently tested with nrf52 core first and port it to samd later one.

PTS93, 2019-04-26:
I did make a case for this a while back for both CircuitPython and Arduino, if you wanna have a look at my reasoning:
https://github.com/adafruit/Adafruit_CircuitPython_HID/issues/25#issuecomment-409014365
https://github.com/arduino/ArduinoCore-samd/issues/339
The Arduino issue mostly just references the comment I made for CircuitPython.

Given that tinyusb is already being used for CircuitPython and that it will now also be available for Arduino would make this the perfect target to integrate this feature and make it available for both platforms.
It would also ensure a somewhat identical behaviour on both platforms which would be great.

Let me know if I can help with it in any way. I have nrf52 boards as well as M0 and M4 boards from Adafruit to test or develop on.

hathach, 2019-04-27:
thanks @PTS93 , now we have a reason to add it. I will try to make a simple example with it, possibly an simple echo script via HID In & Out endpoint using libusb

PTS93, 2019-04-27:
Awesome! That would be tremendous!

hathach, 2019-05-02:
@PTS93 PR #61 should implement this. The hid echo example is https://github.com/hathach/tinyusb/tree/master/examples/device/hid_generic_inout

there is also a simple `hid_test.py` to send and receive from HID device. It is only tested on my ubuntu (macos, windows please check out the py `hid` installation). Let's me know if that meets your expectation. Feel free to discuss/feedback on how it can be enhanced.

PTS93, 2019-05-02:
Awesome! Thank you for getting on this so quickly.
I was able to test it on Ubuntu with an nRF52840 Feather Express and the Arduino example.
I also tested it with node.js on the host side which was also working great!

I also tried it on windows but could not get a communication going. Not sure where the issue is. Only tried it with my node.js test script as the python library seems to be *nix only or atleast does not provide instruction for windows.

PTS93, 2019-05-02:
In terms of how it can be enhanced. I think some more user friendly examples would probably be great (e.g. how to send a simple string from arduino to pc), this is something I should be able to help with, I may have some time next week to work on some more example code for both Arduino and PC.

Other than that I will start incorporating this into an existing piece of software that is currently using Serial to communicate. I will let you know how that goes. It would be a pretty good test as this software is sending back and forth a lot of data and relies on being low latency.

hathach, 2019-05-03:
Yeah, the example is not really great, just a proof of concept for now. Too bad, it doesn't run on windows, but windows user can find other tool with HID write to test with. But I agree we need something simple enough but yet running on windows,macos and linux. I will give it another try later on, got to write bunch of codes just now.

PTS93, 2019-05-03:
I am working with @todbot to get a node.js example on windows going. His library is cross-platform so that is technically a good target for a cross-platform example.
I let you know how it goes.

hathach, 2019-05-03:
thanks, that will be very useful. It feel a bit strange that a web script can do better cross-platform than python :D, hopefully it doesn't need much effort for installation  :)

PTS93, 2019-05-03:
Node.js is pretty great for that actually :)
It creates this dependency file automatically similar to this setup.py file that python projects sometimes have.
As long as you have node.js installed its a matter of calling "npm install" and everything will be downloaded and or compiled for you if there is native code.

PTS93, 2019-05-03:
The maintainer from node-hid points to Teensy's raw hid implementation. Maybe it is worth a look to see what Paul did differently. Maybe it is just a matter of the descriptor that windows doesn't like it the way it is.
https://github.com/PaulStoffregen/cores/blob/master/usb_rawhid
https://github.com/PaulStoffregen/cores/blob/master/usb_rawhid/usb.c

hathach, 2019-05-04:
@PTS93 thank for suggestion. Does node-hid work on Linux but failed on windows, or it doesn't work with current tinyusb at all ?

I need a bit of help to get started with node-hid. I just install nodejs, npm and node-hid on my ubuntu, but I have no idea which command/script to run. Can you give me a simple script that send and receive like python script, and also the command to run it as well ( I am a total noob to node). 

PS: The node about the key logger is interesting, I didn't know it until now :D

PTS93, 2019-05-04:
Yes, it worked on Linux without issues but failed on Windows.
You can find a simple script in the first post of this issue: https://github.com/node-hid/node-hid/issues/307
This is what I tested with on Linux and Windows.

If you need any assistance with node.js let me know :)
Before running the script I posted just run `npm install node-hid` in the folder that you saved the script.
By default, dependencies are saved locally for every script in a "node_modules" sub-folder.
You can also install it globally by using the `-g` flag so `npm install -g node-hid`. Though the common practice is to install them locally so every script can have its correct version of a dependency.

hathach, 2019-05-06:
Ok, I just run the following script
 
```
var HID = require('node-hid');
var devices = HID.devices();
var deviceInfo = devices.find( function(d) {
    var isNRF = d.vendorId===0X239A && d.productId===0X8029;
    return isNRF;
});
if( deviceInfo ) {
	console.log(deviceInfo)
	var device = new HID.HID( deviceInfo.path );
	device.on("data", function(data) {console.log(data)});

	device.on("error", function(err) {console.log(err)});

	setInterval(function () {
		device.write([0x00, 0x01, 0x01, 0x05, 0xff, 0xff]);
	},500)
}
```
output on ubuntu

```
$ node hid.js 
{ vendorId: 51966,
  productId: 16388,
  path: '/dev/hidraw8',
  serialNumber: '123456',
  manufacturer: 'tinyusb.org',
  product: 'tinyusb device',
  release: 256,
  interface: 0,
  usagePage: 14901,
  usage: 11825 }
<Buffer 01 01 05 ff ff>
<Buffer 01 01 05 ff ff>
<Buffer 01 01 05 ff ff>
<Buffer 01 01 05 ff ff>
<Buffer 01 01 05 ff ff>
```

on windows

```
node hid.js 
{ vendorId: 51966,
productId: 16388,
path:'\\\\?\\hid#vid_cafe&pid_4004#6&34b4a546&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}',
serialNumber: '123456',
manufacturer: 'tinyusb.org',
product: 'tinyusb device',
release: 256,
interface: -1,
usagePage: 65280,
usage: 1 } 
```

I will do more testing later on, does the output look similar on your pc @PTS93 , the `<Buffer` is indicator for success transmission right ? I will do more test later on.

PTS93, 2019-05-06:
Correct, the buffer is the buffer that was sent to the MCU and got echoed back.

todbot, 2019-05-06:
Yeah I don't understand what's happening. On Friday I got ahold of a Adafruit Feather nRF52840 Express to test this more extensively.  I've been trying changes both with the TinyUSB `hid_generic_inout` example and the similar `Adafruit_TinyUSB_Arduino` example with an updated TinyUSB.

On Linux, Raw HID reports from TinyUSB is received, but on MacOS and Windows they are not.  Using either `node-hid`, PJRC's "rawhid" executable (hacked to look for TinyUSB vid/pid), I can send data to TinyUSB but cannot receive.

In looking at the descriptors and comparing them to Teensy's RawHID case, I noticed that TinyUSB global maxPacketSize was 64 and the HID report was 64, but the HID endpoints maxPacketSize was 16. I modified `tusb_config.h` to have `CFG_TUD_HID_BUFSIZE` to 64.  But no effect. (Normally I've seen HID reports that fit within maxPacketSize, with the endpoint size matching EP0 maxPacketSize) 

I also modified the HID report `TUD_HID_REPORT_DESC_GENERIC_INOUT` descriptor to match Teenys's RawHID reports, but no effect there either. Teeny reports are received on all OSes but TinyUSB reports are only received on Linux.

I've also tried modifying the HID endpoint addresses in case there was some collision of the low-7-bits of the address causing issues (e.g. maybe the OS is thinking CDC endpoint 0x82 and HID endpoint 0x02 were the same).  No change.

I have also tried various combinations of calling `tud_hid_report()` in a loop, setting reportIds and off-by-one buffer sizes when sending from TinyUSB.  No change.

My hypothesis is that Linux is being more lenient than Windows and MacOS and there's just a small, probably configuration descriptor-related, change that needs to be made for those host OSes to see TinyUSB reports.   I'm at a loss as to what that might be though.

PTS93, 2019-05-07:
Thank you for the extensive debugging, much appreciated!
Pinging @hathach as he is the core developer of tinyusb and also currently looking into why this is happening.

That it is somehow descriptor related would also be my current assumption.

hathach, 2019-05-07:
@todbot for doing extensive test on this. I will try to look closer at this. As @PTS93 said, it could be an misplaced descriptor and Linux is more forgiving than windows/macos.

hathach, 2019-05-24:
@PTS93 PR #76 should fix issue with windows, it is due to incorrect logic check with hid_report len, windows also send max ep size 64, while Linux only send as many as needed.  I added `hid_test.js` for testing out example as well. Though I don't know nodejs, it will help others a lot if you could help to  update the script to be more descriptive and easier for new user. It is absoluately no problems if you are busy, I will figure it out later on :)

PTS93, 2019-05-24:
Awesome! I will see if I can take a look this weekend :)

todbot, 2019-05-24:
Thank you!  I was so close in my hacking around on the tinyusb source to getting this. Dang! :)
I have verified this change by flashing the "hid_generic_inout" example on a "feather_nrf52840_express" board and can send and receive on both MacOS 10.14 and Windows 10 Pro 64-bit, using both `node-hid` and a hacked up version of PJRC's "rawhid.exe" that allows arbitrary VID/PID.

There is one change though in the tester programs.  I have a PR #77 to address this.  Basically, always try to send the same buffer size as the reportLength.

hathach, 2019-05-25:
it is a stupid mistake, quick fix, but I am kept busy with the Arduino library. Thanks for the PR, just merged :)

PTS93, 2019-05-26:
I can also confirm that everything is working fine on windows now :)
I rewrote the example a bit to make it a bit easier to work with for someone in experienced with node.js
I made a PR here https://github.com/hathach/tinyusb/pull/78

PTS93, 2019-05-26:
@todbot maybe you could try this on osx to see if I implemented your fix correctly?

hathach, 2019-06-10:
implemented, should be closed. Please re-open if needed.

closed 2019-06-10 by hathach

dallaswood96, 2022-10-19:
Hello - I know this is an old issue but I seem to be having the same issue as described by todbot on MacOS. I am currently testing this with a very generic python script to read/write HID devices (although have tested with multiple others). This script works on Linux but not on MacOS (I cannot confirm Windows). I can confirm my device is sending the hid report with no issues by using a Beagle USB protocol analyzer. The device in question is using the RP2040 chip. I am at a loss on where to go from here. Has anyone experienced this issue on MacOS? Any advice on where to look?

bohdan-tymkiv, 2023-05-04:
@dallaswood96 @hathach I am experiencing the same issue on Windows 10/11 + STM32G4. Reports sent by the host reach the device, and the device responds to the requests, tud_hid_report() returns success, but these responses (transfers Device -> Host) never reach the host. I am using libhidapi on the Host, and hid_read_timeout() always returns zero (timeout reached). I am completely stuck and have no idea how to proceed further. On Linux host everything works like a charm.

hathach, 2023-05-24:
please open an new dicussion with all the detail of your issue, bumping up an resolved thread is never an idea way.