hathach/tinyusb #62: porting tinyusb to ZYNQ

LucasGit, 2019-05-07:
**Here is my question**

LucasGit, 2019-05-07:
after finish port ,tinyusb can  detect device insert & remove, but when i try to run enumration ,i find that
when first Device DESC was sent ,i can not recv any data and any interrupt will not be triggerd ,so i check the code & debug the REG status,  async enable bit in USBCMD reg was not set when ehci was init  in raw tinyusb project , so i wanna know its a bug or it is  intention ?

LucasGit, 2019-05-07:
after some code modify (Enable Async Schedule when init EHCI module ,and disable Periodic Schedule bit) 
then i find that ASYNCLISTADDR reg will be clear to 0x00000000 and USBSTS reg show Async Schedule is active and Reclamtion bit is 1 (ZYNQ TRM show 1 mean empty async schedule, 0 means unprocessed async transactions)

hathach, 2019-05-09:
Hi, I am currently focus on device stack, the host stack can be a bit unreliable at the moment since it isn't keep up to date. Though, last time I checked the host stack still works with nxp lpc43/18 EHCI. I will try to test it again whenever I could. But I haven't tested it on other mcu at all.

LucasGit, 2019-05-09:
> Hi, I am currently focus on device stack, the host stack can be a bit unreliable at the moment since it isn't keep up to date. Though, last time I checked the host stack still works with nxp lpc43/18 EHCI. I will try to test it again whenever I could. But I haven't tested it on other mcu at all.

@

hathach, 2019-05-09:
@LucasGit did you figure it out ??

LucasGit, 2019-05-09:
sorry , i mishandling close this issue

LucasGit, 2019-05-09:
@hathach and i meet a weird issue now ,when i try to disable Peridic schedule with USBCMD related bit set 0 when init EHCI ,and ASYNCLISTADDR (0x034d312) was init with right value, then with device attach , interrupt was triggered ,but !!! i find that  ASYNCLISTADDR  was change to a weird value  (0x034d312)
and i find that FRAMEINDEX register was changed again and again (not zero) at the enum stage 

so have you ever met this error ? or some other advice about this phenomenon ? and i compare lpc43/18 and zynq 7000 user guide about EHCI , so i do not  find evident different between them !!!

LucasGit, 2019-05-09:
> after some code modify (Enable Async Schedule when init EHCI module ,and disable Periodic Schedule bit)
> then i find that ASYNCLISTADDR reg will be clear to 0x00000000 and USBSTS reg show Async [...]

i solve this problem , cause i comment a ReducePower function ,but i do not have a good explanation about this .
and when solve this , ASYNCLISTADDR  value is right ,but the first control xfer is still not success ,but i find USBSTS value is  0x0000c08  means that Periodic and Async Schedule was run at the same time ? so will periodic schedule run  block async schedule ? and at this condition make control transfer failed ?????
**so   i try to disable periodic schedule ,but i meet a weird problem as the former message mentioned**

LucasGit, 2019-05-09:
> i solve this problem , cause i comment a ReducePower function ,but i do not have a good explanation about this .
> and when solve this , ASYNCLISTADDR value is right ,but the first control xfer is [...]

@hathach

LucasGit, 2019-05-09:
i find that when get first 8 byte device descriptor , the setup request direction is IN ? i read the usb protocol ,it said the dir here should be OUT !!!

LucasGit, 2019-05-09:
How did you test the tinyusb code ? Does it can process entire enumeration  successfully ?

hathach, 2019-05-09:
@LucasGit last time I tested, tinyusb host can mount cdc, msc, hid and hub as well. There should be no problems with protocol core. Although the hcd layer is not well abstract yet. You would need a bit of effort to porting, but in the low level only. For Frameindex reg, it is increased by hw each miro-frame (ehci specs) and it is absolutely normal.

LucasGit, 2019-05-15:
@hathach i check the init process seems ok ,and when if send fitrst setup package , the head QH "current qTD pointer" area do not overwrite with the right qTD pointer address , and then the IOC interrupt was not triggered either, i had checked that ASYNC schedule bit in USBSTS reg is set 1, 
Reclamation bit is 1, (empty async schedule), so have you ever met this condition  ?

LucasGit, 2019-05-20:
@hathach Enumeration failed bug has been solved ,the reason maybe 
1. in ZYNQ platform ,when HCD write data to RAM ,need an extra operation cache flush to make sure data was write from cache to RAM immediately , all RAM operation which will be operated by HCD & usb controller need to flush cache
2. i disable periodic schedule when init EHCI ,and when i enable it , enumeration seems does not work , but  i don't  know why  ,so i have a doubt that how ASYNC & Periodic schedule work together

hathach, 2019-05-20:
@LucasGit thanks for updating the status, regarding the periodic list, I remembered using NXP extended feature for using only 8 bit framelist to save SRAM. Zynq probably doesn't support that, it could be the reason. Unfortunately, I don't have enough time to rework the host stack/API for better portability. I will try to do so later on.

LucasGit, 2019-05-22:
Glad to recv your answer , the periodic problem i will try to solve it later !
I am focus on enumeration test now ,and i find that when the usb mico physical interface is not tie ,and then insert to the HOST, the PORT CONNECTED interrupt will be triggered nearly more than 1 times ,and the second time interrupt , the enumeration will be failed , 
do you have any good solutions to solve this problem ? or this problem is not normal !?

LucasGit, 2019-05-23:
> Glad to recv your answer , the periodic problem i will try to solve it later !
> I am focus on enumeration test now ,and i find that when the usb mico physical interface is not tie ,and then insert to [...]

i find a bug here
file: usbh.c->enum_task（）
code:
//------------- Set new address -------------//
  uint8_t const new_addr = get_new_address();
  TU_ASSERT(new_addr <= CFG_TUSB_HOST_DEVICE_MAX); // TODO notify application we reach max devices

when reach max device address ,the max address will be allocated again ,and then when run open pipe the list_insert() be excuted , the schedule list will be error occured that qh->next_address will point to himself now schedule list head

hathach, 2019-05-23:
@LucasGit ah thanks, apparently I didn't test much with hub :)

LucasGit, 2019-05-24:
I do not solve the problem that when echi_init() [Periodic & Async Schedule are both enableed] ,the enumeration will be failed , i note you comment in ehci_init() that 
`  //------------- USB CMD Register -------------//
  regs->command |= TU_BIT(EHCI_USBCMD_POS_RUN_STOP) | TU_BIT(EHCI_USBCMD_POS_ASYNC_ENABLE)
                | TU_BIT(EHCI_USBCMD_POS_PERIOD_ENABLE) **// TODO enable period list only there is int/iso endpoint**
                | ((EHCI_CFG_FRAMELIST_SIZE_BITS & TU_BIN8(011)) << EHCI_USBCMD_POS_FRAMELIST_SZIE)
                | ((EHCI_CFG_FRAMELIST_SIZE_BITS >> 2) << EHCI_USBCMD_POS_NXP_FRAMELIST_SIZE_MSB);

`
so does it means when periodic is necessary enable it ,and if there is a async schedule we should disable it cause periodic schedule will block async schedule ? 
I puzzle that how to make periodic & async schedule auto switch according to current transmit type!
Cause there maybe such condition : ctrl ,bulk and interrupt transmit exist at the same time , so how to configure the usb ehci  reg to make it finish all the transmit auto ?

LucasGit, 2019-06-24:
1. usb host stack is working fine now ,& the most problem is cache refresh

closed 2019-06-24 by LucasGit

LucasGit, 2019-08-19:
> I do not solve the problem that when echi_init() [Periodic & Async Schedule are both enableed] ,the enumeration will be failed , i note you comment in ehci_init() that
> ` //------------- USB CMD [...]

 I test this in my latest code again, it disappears,,,,,,, Amazing!!! async and periodic work find at the same time ,here is the register status :
              USBCMD: 0x00088b3d 
              USBSTS: 0x0000c088 
            USBINTER: 0x000c0026 
             FRINDEX: 0x0000111a 
       CTRLDSSEGMENT: 0x00000000 
        PERIODICBASE: 0x04d00000 
       ASYNCADDRBASE: 0x04d00120 
          TT_CONTROL: 0x00000000 
          CONFIGFLAG: 0x00000001 
              PORTSC: 0x88001205 
i test the async wich msc class code ,and the periodic i have not implement ,so i just figure it is ok through the USBSTS register