hathach/tinyusb #69: [Bug] for Metro Express boards .data section is not in bin output file

dpgeorge, 2019-05-20:
**Describe the bug**
`.data` section is not placed in output `.bin` file.

**Set up (please complete the following information):**
 - OS: Linux
 - Board: metro_m0_express and metro_m4_express
 - Firmware Code: any

**To Reproduce**
Build either of the above boards.

**Expected behavior**
`.data` section is expected to be in `.bin` file, but is not.

The `samd21g18a_flash.ld` (and 51) link scripts name the data section `.relocate` yet `examples/rules.mk` is using `data` in the rule to create the `.bin`.  I guess these section names should be made to match each other?

hathach, 2019-05-20:
thanks @dpgeorge , it is indeed and a bug, there is a pr #68 by zardam that should fixed this issue. I also added uf2 target to make it easier to work with uf2 boatloader board #70

closed 2019-05-20 by hathach

dpgeorge, 2019-05-20:
Thanks for the fix!  Indeed I was working with uf2 files, so it's nice to have full support for that now in this repo.

hathach, 2019-05-20:
I am glad that helps, feel free to open other issue if needed ;)

PR #70 (2019-05-20): add uf2 target for uf2 bootloader boards
- follow up to #68  and should fix #69 
- metro m0/m4 and feather nrf52840
- only add uf2 as submodule at tools/uf2