hathach/tinyusb #71: [Feature] Uf2 example

hathach, 2019-05-20:
A generic uf2 example that implement [uf2 specs](https://github.com/Microsoft/uf2) for reference. For simplicity, example doesn't do actual flashing, but only print out the addres + binary contents.

cr1901, 2019-10-03:
Thought while it's on my mind.

It's probably difficult to make this portable, but I think a UF2 example should include a fake "bootloader" that prints out the address + binary contents, and then jumps to a fake application in another part of flash that does nothing. There would be a linker script for bootloader and application.

This would require flash write/erase code, changing the vector table location (because bootloader and application would have a set of vectors), and jumping to reset in the board support package.

arturo182, 2020-01-04:
Finally pushed my initial implementation of the uf2 bootloader, tested and working on my MIMXRT1011 Feather, needs lots of work but the basic bring up is there: https://github.com/arturo182/tinyusb/tree/uf2

The FAT code is based on the samd uf2 bootloader code.
I tried to make the bootloader itself as platform-agnostic as possible, the flashing functionality is all in the bsp. This is so-so because a lot of it is platform-specific so all boards that use the imxrt family should share it.

It feels like it might be out of scope for tinyusb and maybe it would be better to start the project as a separate codebase and use tinyusb as a module in there but as a proof of concept, it was easier to make it a tinyusb example and take advantage of the build system :)

Lots of work to be done, I also want to implement the uf2 hid interface as the msc is not always the best way to mass-flash a lot of boards.

arturo182, 2020-01-04:
Ping @tannewt cause I know you've been wanting to check this out :sweat_smile:

tannewt, 2020-01-05:
Woohoo! Thank you! 🎉 I think we could make a TinyUF2 repo for it with TinyUSB as a submodule. I'm working on logic analyzer code in the same way.

hathach, 2020-01-17:
Sorry, I am late to the party, yeah, full feature UF2 is too much for this repo, it should be best as a separated project. All TinyUSB's example is meant to be simple and proof of concept.

arturo182, 2020-01-17:
Yeah, I started work on a separate repo for this :) https://github.com/arturo182/tinyuf2

hathach, 2020-01-17:
> Yeah, I started work on a separate repo for this :) https://github.com/arturo182/tinyuf2

superb !!!

hathach, 2020-11-05:
closed since there is a dedicated repo for this purpose already.

closed 2020-11-05 by hathach