hathach/tinyusb #72: [Bug] Computing time intervals with millis

dpgeorge, 2019-05-20:
**Describe the bug**
When computing time intervals care must be taken with wrap around.  Eg:
```c
static inline void board_delay(uint32_t ms)
{
  uint32_t start_ms = board_millis();
  while( board_millis() < start_ms + ms) {}
}
```
If `start_ms` is close to `0xffffffff` such that adding `ms` wraps it around then the function will most likely return straightaway.

hathach, 2019-05-21:
thanks again, PR #73 should fix this. Feel free to re-open & comment if you think it doesn't

closed 2019-05-21 by hathach

dpgeorge, 2019-05-21:
A simple way to fix it, due to 2's complement arithmetic overflow which is well-defined behaviour (at least in C99), is like this:
```c
static inline void board_delay(uint32_t ms)
{
  uint32_t start_ms = board_millis();
  while( board_millis() - start_ms < ms) {}
}
```

Also, there are many places in the code that use this pattern, eg in `examples/device/cdc_msc_hid/src/main.c:hid_task()`:
```c
if ( board_millis() < start_ms + interval_ms) return;
```

hathach, 2019-05-21:
Ah thank you, that is very neat :smile: . That proves how amazing Math is :1234:

commit 73c7db38 (2019-05-21): fix #72 address overflow with board_delay
commit 4ab392cc (2019-05-21): Merge pull request #73 from hathach/develop

fix #72 address overflow with board_delay