hathach/tinyusb #83: Wrong macro name for VBUS enable on STM32F412[Bug]

hierophect, 2019-07-03:
**Describe the bug**
I'm attempting to use TinyUSB with an STM32F412ZG. `dcd_stm32F4.c `uses the `USB_OTG_GCCFG_VBUSBSEN` flag for usb power detection. However, it looks like the F412 has different names set up for the `USB_OTG_GCCFG` register that stores this information; I had to replace this with `USB_OTG_GCCFG_VBDEN`. 

Should this be made conditional? It's the only incompatibility I've encountered thus far, and I'm not intimately familiar with why this name change exists between the F412 and other F4s like the F411.  

**Set up (please complete the following information):**
Board: NUCLEO-F412ZG 
MCU: STM32F412ZG
file: portable/st/stm32f4/dcd_stm32f4.c

**To Reproduce**
Occurs on build.

**Expected behavior**
```
../../lib/tinyusb/src/portable/st/stm32f4/dcd_stm32f4.c:163:24: error: 'USB_OTG_GCCFG_VBUSBSEN' undeclared (first use in this function); did you mean 'USB_OTG_GCCFG_VBDEN'?
```

hathach, 2019-07-05:
Hi, can you upload nucleo-f412zg board files here (or better submit PR containing it), so that I could run make and reproduce it here, making sure it fixed issue for all current board. 

The pr doesn't need to be well tested, you can update it later on.

hierophect, 2019-07-18:
What would be the format you need for a PR? A bsp+example combo? 

My fork of Circuitpython will have this issue on compile:
https://github.com/hierophect/circuitpython/tree/F4xx-port-setup/lib

hathach, 2019-07-19:
@hierophect no need to worry about this, I made a changes already, please check to see if it works for your board.
https://github.com/hathach/tinyusb/commit/ed6fa460e9a885bb84a6f1588763ea5fb3f4df49

hathach, 2019-07-19:
and don't worry about the F411 discovery bsp files, I will add them once the board (on-order) is delivered to me

hierophect, 2019-07-19:
All set. Thanks for the fix.

closed 2019-07-19 by hierophect