hathach/tinyusb #88: [Q&A] Respond to Feature Reports?

jeremywood, 2019-07-24:
I'm trying to handle feature reports, but it doesn't look like that's implemented. Or am I missing something? Thank you!

hathach, 2019-07-24:
Could you please provide more information, your setup: os, board, which example to run. Sets to reproduce it, what to expect and what doesn't work.

jeremywood, 2019-07-24:
@hathach Thanks for the speedy response! I've realized that this was an error on my part in understanding the client application and what the actual error was. What was actually happening is that no size was being specified in the HID descriptor for the feature report, and the client application was trying to allocate a zero-sized buffer 😬.

What I needed to do on the TinyUSB side was to define a custom descriptor with a HID_FEATURE with HID_REPORT_COUNT defined.

Here's what that descriptor ended up looking like, if this kind of issue happens to someone else:

```
#define TUD_HID_REPORT_DESC_CUSTOM(report_size, ...) \
    HID_USAGE_PAGE_N ( HID_USAGE_PAGE_VENDOR, 2   ),\
    HID_USAGE        ( 0xA0                       ),\
    HID_COLLECTION   ( HID_COLLECTION_APPLICATION ),\
      /* Report ID if any */\
      __VA_ARGS__ \
      /* Input */ \
      HID_USAGE        ( 0xA1                                   ),\
      HID_LOGICAL_MIN  ( 0x00                                   ),\
      HID_LOGICAL_MAX_N( 0xff, 2                                ),\
      HID_REPORT_SIZE  ( 8                                      ),\
      HID_REPORT_COUNT ( 8                                      ),\
      HID_INPUT        ( HID_DATA | HID_VARIABLE | HID_ABSOLUTE | HID_WRAP_NO | HID_LINEAR |HID_PREFERRED_STATE | HID_NO_NULL_POSITION ),\
      HID_USAGE        ( 0xa3                                    ),\
      HID_REPORT_COUNT ( 24                                      ),\
      HID_FEATURE      ( HID_DATA | HID_VARIABLE | HID_ABSOLUTE | HID_WRAP_NO | HID_LINEAR |HID_PREFERRED_STATE | HID_NO_NULL_POSITION | HID_NON_VOLATILE),\
    HID_COLLECTION_END \
```

closed 2019-07-24 by jeremywood

hathach, 2019-07-26:
thank you very much @jeremywood, indeed the current examples doesn't have a template for feature report. I will try to a way to add them in later, though since HID descriptor is quite complicated and application specific, It probably won't cover all use cases. Though at least application can input its own descriptor. Let's me know if you need any help