hathach/tinyusb #96: [Q&A]data buffer err in qhd make usb controller dead in zynq7000 platform

LucasGit, 2019-08-12:
**Here is my question**
when i test the usb host code with msc class ,and use the fatfs stack write data to the usb stick
when test write arbitrary size data the host controller seemed deal with the reclamation bit always 1
but the IOC not be triggered never

typical log is here:
f_write return code:1, 8865(espect size), 409(actual write size)

register status
              USBCMD: 0x00000b21 
              USBSTS: 0x0000a088 
            USBINTER: 0x000c0026 
             FRINDEX: 0x00002a1f 
       CTRLDSSEGMENT: 0x00000000 
        PERIODICBASE: 0x04cfe000 
          TT_CONTROL: 0x00000000 
          CONFIGFLAG: 0x00000001 
              PORTSC: 0x88001205 

qhd list print (error bit is marked with bold style):

--ASYNC Head 
[QH Node] 0x0x4cfe120 --> 
QH 0x0x4cfe120
DWord 0: QH Horizontal Pointer 0x0x4cfe220, Type 1 , T 0 
DWord 1: RL 5, C 0, MaxPckLen 8, H 1, DTC 1, EPS 2, EndPt 0,I 0, DevAddr 0 
DWord 2: Mult 1,PortNum  0, HubAddr 0, C-Mask 0, S-Mask 0 
DWord 3: Cur qTD Pointer 0x0x4cfe160
qTD Overlay 
DWord 0: Nxt 	qTD Pointer 0x0x0, T 1 
DWord 1: Alt Nxt qTD Pointer 0x0x20, T 1 
DWord 2: DT 0, Total bytes 0, IOC 1, C_Page 0, Cerr 3, PID 1, 
 active 0, halted 0, buffer 0,babble err 0, xact err 0, non-hs-missed_uframe 0, non hs split stat 0, Ping err 1 
DWord 3: Buffer 0: 0x0 
DWord 4: Buffer 1: 0x1002 
DWord 5: Buffer 2: 0x2002 
DWord 6: Buffer 3: 0x3000 
DWord 7: Buffer 4: 0x4000 
@ Extra Info - 
used 1,removing 0,pid 0,interval_ms 0,total_xferred_bytes_ms 0 
p_qtd_list_head 0x0x0,p_qtd_list_tail 0x0x0
@ qtd list print - 
[QH Node] 0x0x4cfe220 --> 
QH 0x0x4cfe220
DWord 0: QH Horizontal Pointer 0x0x4cfe1e0, Type 1 , T 0 
DWord 1: RL 5, C 0, MaxPckLen 512, H 0, DTC 0, EPS 2, EndPt 2,I 0, DevAddr 1 
DWord 2: Mult 1,PortNum  0, HubAddr 0, C-Mask 0, S-Mask 0 
DWord 3: Cur qTD Pointer 0x0x4cfe280
qTD Overlay 
DWord 0: Nxt 	qTD Pointer 0x0x0, T 1 
DWord 1: Alt Nxt qTD Pointer 0x0x4000020, T 1 
DWord 2: DT 1, Total bytes 512, IOC 0, C_Page 0, Cerr 3, PID 0, 
 active 1, halted 0, **buffer 1**,babble err 0, xact err 0, non-hs-missed_uframe 0, non hs split stat 0, Ping err 0 
DWord 3: Buffer 0: 0x9bc399 
DWord 4: Buffer 1: 0x9bd010 
DWord 5: Buffer 2: 0x9be018 
DWord 6: Buffer 3: 0x9bf000 
DWord 7: Buffer 4: 0x9c0000 
@ Extra Info - 
used 1,removing 0,pid 0,interval_ms 0,total_xferred_bytes_ms 10562 
p_qtd_list_head 0x0x4cfe260,p_qtd_list_tail 0x0x4cfe280
@ qtd list print - 
DWord 0: Nxt 	qTD Pointer 0x0x4cfe280, T 0 
DWord 1: Alt Nxt qTD Pointer 0x0x1f0020, T 1 
DWord 2: DT 0, Total bytes 0, IOC 0, C_Page 0, Cerr 3, PID 0, 
 active 0, halted 0, buffer 0,babble err 0, xact err 0, non-hs-missed_uframe 0, non hs split stat 0, Ping err 0 
DWord 3: Buffer 0: 0xa12048 
DWord 4: Buffer 1: 0xa13000 
DWord 5: Buffer 2: 0xa14000 
DWord 6: Buffer 3: 0xa15000 
DWord 7: Buffer 4: 0xa16000 
DWord 0: Nxt 	qTD Pointer 0x0x0, T 1 
DWord 1: Alt Nxt qTD Pointer 0x0x4000020, T 1 
DWord 2: DT 0, Total bytes 1024, IOC 0, C_Page 0, Cerr 3, PID 0, 
 active 1, halted 0, buffer 0,babble err 0, xact err 0, non-hs-missed_uframe 0, non hs split stat 0, Ping err 0 
DWord 3: Buffer 0: 0x9bc199 
DWord 4: Buffer 1: 0x9bd000 
DWord 5: Buffer 2: 0x9be000 
DWord 6: Buffer 3: 0x9bf000 
DWord 7: Buffer 4: 0x9c0000 
[QH Node] 0x0x4cfe1e0 --> 
QH 0x0x4cfe1e0
DWord 0: QH Horizontal Pointer 0x0x4cfe180, Type 1 , T 0 
DWord 1: RL 5, C 0, MaxPckLen 512, H 0, DTC 0, EPS 2, EndPt 1,I 0, DevAddr 1 
DWord 2: Mult 1,PortNum  0, HubAddr 0, C-Mask 0, S-Mask 0 
DWord 3: Cur qTD Pointer 0x0x4cfe2a0
qTD Overlay 
DWord 0: Nxt 	qTD Pointer 0x0x0, T 1 
DWord 1: Alt Nxt qTD Pointer 0x0xd0020, T 1 
DWord 2: DT 0, Total bytes 13, IOC 1, C_Page 0, Cerr 3, PID 1, 
 active 1, halted 0, buffer 0,babble err 0, xact err 0, non-hs-missed_uframe 0, non hs split stat 0, Ping err 0 
DWord 3: Buffer 0: 0xa12048 
DWord 4: Buffer 1: 0xa13010 
DWord 5: Buffer 2: 0xa14018 
DWord 6: Buffer 3: 0xa15000 
DWord 7: Buffer 4: 0xa16000 
@ Extra Info - 
used 1,removing 0,pid 1,interval_ms 0,total_xferred_bytes_ms 0 
p_qtd_list_head 0x0x4cfe2a0,p_qtd_list_tail 0x0x4cfe2a0
@ qtd list print - 
DWord 0: Nxt 	qTD Pointer 0x0x0, T 1 
DWord 1: Alt Nxt qTD Pointer 0x0xd0020, T 1 
DWord 2: DT 0, Total bytes 13, IOC 1, C_Page 0, Cerr 3, PID 1, 
 active 1, halted 0, buffer 0,babble err 0, xact err 0, non-hs-missed_uframe 0, non hs split stat 0, Ping err 0 
DWord 3: Buffer 0: 0xa12048 
DWord 4: Buffer 1: 0xa13000 
DWord 5: Buffer 2: 0xa14000 
DWord 6: Buffer 3: 0xa15000 
DWord 7: Buffer 4: 0xa16000 
[QH Node] 0x0x4cfe180 --> 
QH 0x0x4cfe180
DWord 0: QH Horizontal Pointer 0x0x4cfe120, Type 1 , T 0 
DWord 1: RL 5, C 0, MaxPckLen 64, H 0, DTC 1, EPS 2, EndPt 0,I 0, DevAddr 1 
DWord 2: Mult 1,PortNum  0, HubAddr 0, C-Mask 0, S-Mask 0 
DWord 3: Cur qTD Pointer 0x0x4cfe1c0
qTD Overlay 
DWord 0: Nxt 	qTD Pointer 0x0x4cfe1c0, T 0 
DWord 1: Alt Nxt qTD Pointer 0x0x10020, T 1 
DWord 2: DT 0, Total bytes 0, IOC 1, C_Page 0, Cerr 3, PID 1, 
 active 0, halted 0, buffer 0,babble err 0, xact err 0, non-hs-missed_uframe 0, non hs split stat 0, Ping err 1 
DWord 3: Buffer 0: 0xa12065 
DWord 4: Buffer 1: 0xa13001 
DWord 5: Buffer 2: 0xa1400b 
DWord 6: Buffer 3: 0xa15000 
DWord 7: Buffer 4: 0xa16000 
@ Extra Info - 
used 1,removing 0,pid 0,interval_ms 0,total_xferred_bytes_ms 0 
p_qtd_list_head 0x0x0,p_qtd_list_tail 0x0x0
@ qtd list print - 
[QH Node] 0x0x4cfe120 --> 
--End

and I try to search some solutions on the internet ,but has no result
so i dont know its usb controller's quetion or the fatfs's qustion ????
have you ever met this ?

LucasGit, 2019-08-14:
I test msc write interface with a random buffer address (total buffer size is 20KB ,and the address is assigned to qtd buffer address) and data size is 512 bytes algin (total size is 8KB limit) ,and then write the data into a fix secotor address (which i am sure is valid to write) , as result i found the data buffer error occur easily

then i get some some advice that  zynq 7000 usb controller has it's own DMA controller and when DMA controller do data transfer ,4 bytes align is recommanded, so i 

try to fix the data buffer address with random 4 bytes algin value ,all seems fine ,more stable than before

as a result i found that fatfs f_write can not make sure the buffer address which transfer to the msc interface is 4 bytes algined, so i think there may be some quetions when usb controller DMA hanle a RAM space which buffer address is  not 4 bytes aligned

hathach, 2019-08-14:
Since I don't have any way to reproduce the issue, the zynq7000 is also new to me as well. I am afraid you are pretty much on your own to resolve this.

LucasGit, 2019-08-14:
with the 4 bytes aligned workaround , the code work fine very well @hathach

closed 2019-08-14 by LucasGit