hathach/tinyusb #102: Port NXP LPC55S6x

hathach, 2019-08-20:
**Porting layer**
 - [x] Device Controller Driver (DCD)
 - [ ] Host Controller Driver (HCD)
 - [x] Board Supported Package (BSP)
 - [ ] OS Abstraction Layer (OSAL)

**Description**
This new mcu share same usb controller with already ported lpc, it should be quick to get supported.

tannewt, 2019-08-20:
Does this chip have High Speed USB as well?

hathach, 2019-08-21:
@tannewt yeah the usb device controller in this chip is similar to one in lpc18xx/43xx which is dual usb with highspeed mode. However, only port0 can run at HS without external PHY, port1 need external component to run a HS otherwise it is FS. Sumup 1 HS + 1 FS (without external component).

commit 4fe34c8f (2019-09-08): Merge pull request #120 from hathach/develop

Ported STM32F412 Discovery board, close #102
closed 2019-09-08 by hathach