hathach/tinyusb #103: Port STM32F412 Discovery board

hathach, 2019-08-20:
**Porting layer**
 - [ ] Device Controller Driver (DCD)
 - [ ] Host Controller Driver (HCD)
 - [x] Board Supported Package (BSP)
 - [ ] OS Abstraction Layer (OSAL)

ladyada, 2019-09-06:
@hathach please try to finish this next since we have circuitpython working other than USB so this is the only step :)

hathach, 2019-09-07:
@ladyada yeah, got the board blinky, but the usb somehow isn't regconzied. troubleshooting it asap.

hathach, 2019-09-08:
closed by #120

closed 2019-09-08 by hathach

ladyada, 2019-09-09:
i tried with great success - here's the bin from thach

be sure to install a jumper into the **JP3** USB jumper, its near the green **PWR** LED
[stm32f412disco.zip](https://github.com/hathach/tinyusb/files/3588803/stm32f412disco.zip)