hathach/tinyusb #115: WebUSB Device Example fails to find a driver automatically on Windows 7

cr1901, 2019-09-05:
**Describe the bug**
From what I can gather [looking at](https://github.com/hathach/tinyusb/blob/master/examples/device/webusb_serial/src/usb_descriptors.c#L170) the WebUSB example, running the example on Windows should associate the WebUSB interface with the WinUSB driver _automatically and without user intervention_. Windows fails to find a driver on my machine.

**Set up (please complete the following information):**
 - OS: Windows 7
 - Board: stm32h743nucleo
 - Firmware Code: examples/device/webusb_serial

**To Reproduce**
Steps to reproduce the behavior:
1. Obtain a machine running Windows 7.
2. Compile the `webusb_serial` example. In my case, I used: `make -C examples/device/webusb_serial BOARD=stm32h743nucleo`.
3. Wait for Windows as it attempts to load a drivers for each function,
4. WinUSB driver will fail to load for the WebUSB interface.

**Expected behavior**
Windows would find the magic "WinUSB" descriptors and automatically load the WinUSB driver.

**Screenshots**
[image]

**Additional context**
On Windows 7, the CDC driver is not normally loaded automatically; I used the provided [inf file](https://github.com/hathach/tinyusb/blob/master/tools/windows_cdc_driver/tusb_usbser.inf), and loading the driver has been automatic ever since. Perhaps the same should be done for vendor devices _as a workaround_?

todbot, 2019-09-05:
WebUSB is not supported natively on Windows 7, as I understand it.  See: https://developers.google.com/web/updates/2016/03/access-usb-devices-on-the-web

hathach, 2019-09-05:
@cr1901 it is not a bug, windows 7 doesn't support automatically binding winUSB driver, you need to use tool such as [zadiag](https://zadig.akeo.ie/) to install winUSB driver for the interface manually. It will work afterwards.

cr1901, 2019-09-05:
@hathach @todbot 

I was able to get WinUSB to work using the following `.inf` file I created (_please do not actually use as-is for general use_):

```
;
;
; Installs WinUsb
;

[Version]
Signature = "$Windows NT$"
Class     = USBDevice
ClassGUID = {88BAE032-5A81-49f0-BC3D-A4FF138216D6}
Provider  = %ManufacturerName%
CatalogFile = WinUSBInstallation.cat
DriverVer=09/04/2012,13.54.20.543

; ========== Manufacturer/Models sections ===========

[Manufacturer]
%ManufacturerName% = Standard,NTamd64

[Standard.NTamd64]
%DeviceName% =USB_Install, USB\VID_CAFE&PID_4011&MI_02

; ========== Class definition (for Windows 8 and ealier versions)===========

[ClassInstall32]
AddReg = ClassInstall_AddReg

[ClassInstall_AddReg]
HKR,,,,%ClassName%
HKR,,NoInstallClass,,1
HKR,,IconPath,%REG_MULTI_SZ%,"%systemroot%\system32\setupapi.dll,-20"
HKR,,LowerLogoVersion,,5.2

; =================== Installation ===================

[USB_Install]
Include = winusb.inf
Needs   = WINUSB.NT

[USB_Install.Services]
Include =winusb.inf
Needs   = WINUSB.NT.Services

[USB_Install.HW]
AddReg=Dev_AddReg

[USB_Install.Wdf]
KmdfService=WINUSB, WinUsb_Install

[WinUsb_Install]
KmdfLibraryVersion=1.11

[Dev_AddReg]
HKR,,DeviceInterfaceGUIDs,0x10000,"{661c6dfe-865a-4b81-b26a-b9976f64cef8}"

; [DestinationDirs]
; If your INF needs to copy files, you must not use the DefaultDestDir directive here.
; You must explicitly reference all file-list-section names in this section.

; =================== Strings ===================

[Strings]
ManufacturerName="TinyUSB"
ClassName="Universal Serial Bus devices"
DeviceName="TinyUSB Vendor Interface"
REG_MULTI_SZ = 0x00010000
```

Because I run Windows 7 in test mode, I was able to install the WinUSB driver _and_ verify that the demo works, after a nice big red warning:
[image]

I would like to know if this `inf` file works at all on Windows 7 _not_ in test mode. Anyone got a non-test-mode machine and some boards to test :)?

@hathach In your experience, does the [CDC inf file](https://github.com/hathach/tinyusb/blob/master/tools/windows_cdc_driver/tusb_usbser.inf) work on 64-bit Windows 7/8.x (no test mode)?

hathach, 2019-09-05:
for any unsigned INF on windows 7, you need to disable `windows driver signature enforcement` to be able to install. It is one-time work, once it is done and loaded, it should work afterwards. You can just use Zadiag tool, it is much easier :)

closed 2019-09-05 by cr1901