hathach/tinyusb #119: Port iMX RT1060

hathach, 2019-09-06:
**Porting layer**
 - [ ] Device Controller Driver (DCD)
 - [ ] Host Controller Driver (HCD)
 - [ ] Board Supported Package (BSP)
 - [ ] OS Abstraction Layer (OSAL)

**Description**
https://www.nxp.com/products/processors-and-microcontrollers/arm-microcontrollers/i.mx-rt-series/i.mx-rt1060-crossover-processor-with-arm-cortex-m7-core:i.MX-RT1060

tannewt, 2019-11-03:
I see Device and BSP are checked off. Is this checked in? I'd love to start getting CircuitPython going on the iMX devices.

hathach, 2019-11-03:
@tannewt ah not yet, I haven't started any works with iMX devices just yet. The checklist is kind of of work to do, which admittedly is very confusing. It it part of porting template, I should remove other though.

arturo182, 2019-11-09:
I'm looking into this right now.

hathach, 2019-11-09:
@arturo182 great, I am planning to work on this a bit next week as well. Let’s me know if you figure out something

arturo182, 2019-11-10:
https://github.com/arturo182/tinyusb/tree/mimxrt10xx

I made the BSP for mimxrt1010-evk and a dcd for the rt10xx family, i only have the rt1010 evk so I can't test on the others, but I assume the whole family shares the peripheral, just the bigger ones have more instances.

Anyway, I tested and the HID composite example works, i press the button and my mouse moves and the keyboard sends "a", but the cdc_msc is being cranky:

```
[558138.525242] usb 1-1: new high-speed USB device number 19 using xhci_hcd
[558138.668883] usb 1-1: config 1 interface 1 altsetting 0 bulk endpoint 0x2 has invalid maxpacket 64
[558138.668889] usb 1-1: config 1 interface 1 altsetting 0 bulk endpoint 0x82 has invalid maxpacket 64
[558138.668893] usb 1-1: config 1 interface 2 altsetting 0 bulk endpoint 0x3 has invalid maxpacket 64
[558138.668897] usb 1-1: config 1 interface 2 altsetting 0 bulk endpoint 0x83 has invalid maxpacket 64
[558138.669272] usb 1-1: New USB device found, idVendor=cafe, idProduct=4003, bcdDevice= 1.00
[558138.669277] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[558138.669281] usb 1-1: Product: TinyUSB Device
[558138.669285] usb 1-1: Manufacturer: TinyUSB
[558138.669287] usb 1-1: SerialNumber: 123456
[558138.672832] cdc_acm 1-1:1.0: ttyACM1: USB ACM device
[558138.673523] usb-storage 1-1:1.2: USB Mass Storage device detected
[558138.673895] scsi host1: usb-storage 1-1:1.2
[558139.692458] scsi host1: scsi scan: INQUIRY result too short (5), using 36
[558139.692469] scsi 1:0:0:0: Direct-Access     TinyUSB  Mass Storage     1.0  PQ: 0 ANSI: 2
[558139.693315] sd 1:0:0:0: Attached scsi generic sg1 type 0
[558139.693749] sd 1:0:0:0: [sdb] 16 512-byte logical blocks: (8.19 kB/8.00 KiB)
[558139.693962] sd 1:0:0:0: [sdb] Write Protect is off
[558139.693967] sd 1:0:0:0: [sdb] Mode Sense: 03 00 00 00
[558139.694161] sd 1:0:0:0: [sdb] No Caching mode page found
[558139.694165] sd 1:0:0:0: [sdb] Assuming drive cache: write through
[558139.822249] usb 1-1: reset high-speed USB device number 19 using xhci_hcd
[558139.964881] cdc_acm 1-1:1.0: ttyACM1: USB ACM device
[558140.109021] usb 1-1: reset high-speed USB device number 19 using xhci_hcd
[558140.251244] cdc_acm 1-1:1.0: ttyACM1: USB ACM device
[558140.389029] usb 1-1: reset high-speed USB device number 19 using xhci_hcd
[558140.532068] cdc_acm 1-1:1.0: ttyACM1: USB ACM device
[558140.672287] usb 1-1: reset high-speed USB device number 19 using xhci_hcd
[558140.815229] cdc_acm 1-1:1.0: ttyACM1: USB ACM device
[558140.955609] usb 1-1: reset high-speed USB device number 19 using xhci_hcd
[558141.101391] cdc_acm 1-1:1.0: ttyACM1: USB ACM device
[558141.239045] usb 1-1: reset high-speed USB device number 19 using xhci_hcd
[558141.382352] cdc_acm 1-1:1.0: ttyACM1: USB ACM device
[558141.382923] sd 1:0:0:0: [sdb] Unaligned partial completion (resid=4032, sector_sz=512)
[558141.382936] sd 1:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK
[558141.382941] sd 1:0:0:0: [sdb] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00
[558141.382945] print_req_error: I/O error, dev sdb, sector 0 flags 0
[558141.382953] Buffer I/O error on dev sdb, logical block 0, async page read
```

Out of ideas for now, but wanted to share the progress.

Next week I also want to look into porting the uf2 bootloader to the rt10xx family.

hathach, 2019-11-10:
@arturo182 great news, though since nxp rt series use the the same usb ip known as ip3511. We should just modify the current dcd_lpc_ip3511 to get it running on the rt series as well.

arturo182, 2019-11-10:
I considered it but wasn't sure how that would work with `CHIP_FAMILY` and it will end up quite a ifdef-heavy code, but if you want I can modify the code so it reuses the lpc.

hathach, 2019-11-11:
`CHIP_FAMILY` is just way to find dcd driver file, I will do a rename later to something like `DCD_DRIVER` or similar. The current ip3511 already supported lpc 11, 13, 15, 18, 43 chip, it is generalized enough. Adding mx RT wouldn't make it any worse. If you are not familiar with LPC chips, just all the #ifdef you needed, I will do an abstract work later on :)

hathach, 2019-11-11:
@arturo182 I just update/re-org/rename submodule lpc_driver to nxp_driver and added imx rt mcu driver https://github.com/hathach/nxp_driver/tree/master/sdk/devices . I hope this will help you out with the porting. Let's me know if you need any further helps with porting 1010 evk. I will also try to work with 1064/1060 evk on my end as well.

hathach, 2019-11-22:
@arturo182 I push an PR #220, that the the complete port for iMR RT1064 tested on 1064 EVK board. I haven't looked at other chip family. Hopefully that doesn't have too much of difference, I added the list of board port that I have to the checklist, the only port that I am missing is rt1010evk due to the US export restriction. Hopefully you could find that PR helpful.

arturo182, 2019-11-22:
I have a 1010 evk so can verify it for you

hathach, 2019-11-22:
> I have a 1010 evk so can verify it for you

Thank you very much. Words of warnings: there are lots of changes to refactor the dcd code from lpc family to support both series 🤗🤗

hathach, 2019-11-23:
@arturo182 thanks to great work with rt1010evk, I could easily get these evk up and blinky quickly enough. Since I have been adding quite a bit of RT10xx EVKs, surprisingly they are actually quite the same (initially thought there is probably lots of difference like samd21 & 51). To save your time and effort to merge with tons of changes from the master (due to the refactoring of dcd).  I just go ahead and added the ready-to-run rt1010 evk as well #224. Please give it a try whenever you have time to see if those actually works on the actual hardware, and submit PRs for correction/fix if you see needed. Thank you, hopefully we could close this issue soon enough :)

arturo182, 2019-11-23:
Oh, I was doing that literally just now, bit of a wasted overlapping work :/

hathach, 2019-11-23:
> Oh, I was doing that literally just now, bit of a wasted overlapping work :/

Ooops sorry, that is a bit of unfortunate, I was initially waiting for your work. But then think it is similar enough to 1015 evk, and thought I could just try to  save you some time there. :)

arturo182, 2019-11-23:
Guess I'll just trash my changes and test your PR. Have you tested all the examples with the other EVKs and they all worked? Cause with my previous branch I had a problem with getting MSC to work.

hathach, 2019-11-23:
@arturo182 yeah, I tested with all other EVK, it works rather well, I am sure it is probably something minor. I didn't do anything dramatically with dcd driver, except to generalize it so that it could work with both LPC & miMXRT series. They use the same USB IP after all

arturo182, 2019-11-23:
Ok, I tested with the hid_composite and cdc_msc examples and both work :tada: 

No problem, I'm happy my PR helped you with getting the code up faster, now I can add the tinyusb functionality to my IMXRT10xx port of CP (right now I use LPUART for REPL and have no MSC), I'm gonna push that as a PR to that repo soon too :)

hathach, 2019-11-23:
@arturo182 yeah, the rt mcu is flashless, I spend a several hours and still haven't known how to program it with compatible command line daplink. I gave up and just use the copy bin file as flash target :( . Your board.c and board.mk help a lots to get started with the evk. Thanks again. I think we can safely close this now :)

Let's me know or tag me if you need any help from me doing the CP port for RT10xx, I am happy to help :). Note that the with RT10xx tinyusb doesn't directly implement the `USB_IRQHandler`, application must call `tud_isr()` of device stack. As  #35 describe, some of the RTOSes reserves the direct access to IRQ function, I just starting to role it out when doing with new dcd.

hathach, 2019-11-23:
All changes are merged into master. RT10xx is fully supported. Thank you very much 🤗🤗

closed 2019-11-23 by hathach