hathach/tinyusb #126: Port STM32 L4

hathach, 2019-09-10:
**Porting layer**
 - [x] Device Controller Driver (DCD)
 - [ ] Host Controller Driver (HCD)
 - [x] Board Supported Package (BSP)
 - [ ] OS Abstraction Layer (OSAL)

**Description**
A clear and concise description of what you want to happen.

hathach, 2019-10-03:
tried but couldn't get stm32L476 discovery board running with USB in crystal less mode. Further study is needed.

pigrew, 2019-10-03:
I see a note in the manual about having to enable power gating for USB? Post your code (though I don't have a L476...)?

> After reset, the USB features supplied by VDDUSB are logically and electrically isolated and
therefore are not available. The isolation must be removed before using the USB OTG
peripheral, by setting the USV bit in the PWR_CR2 register, once the VDDUSB supply is
present.

hathach, 2019-10-03:
> I see a note in the manual about having to enable power gating for USB? Post your code (though I don't have a L476...)?

Thanks for the note, I have struggled with L4 for sometime. Got to move on with other issues (too many already :D) . Will come back to test this later on :)

hathach, 2020-05-07:
maybe fixed by #391

kasjer, 2020-05-07:
> maybe fixed by #391

Unfortunately it will not help.

Here is what helped me to get started with enumeration:
```c
    USB_OTG_FS->GCCFG |= USB_OTG_GCCFG_PWRDWN;

    USB_OTG_FS->GUSBCFG &= ~USB_OTG_GUSBCFG_FHMOD;
    USB_OTG_FS->GUSBCFG |= USB_OTG_GUSBCFG_FDMOD;

    USB_OTG_FS->GOTGCTL |= USB_OTG_GOTGCTL_BVALOEN;
    USB_OTG_FS->GOTGCTL |= USB_OTG_GOTGCTL_BVALOVAL;
```

But that is not enough synopsys differs more from what is in F4 chips.
Funny thing is that with just this enumeration succeeded for cdc_msc but only msc worked with some huge delay.
While CDC did not work.
It is caused (maybe among other reasons) by EP0 transfer end interrupt triggered at the end of setup packed, before setup interrupt shows up. With this extra interrupt synopsys driver tries to handle end of transfer interrupt when it did not schedule any transfer yet using transfer data from previous transfer.

hathach, 2020-05-07:
@kasjer that is very helpful finding, I was totally clueless with L4. I may be able to troubleshoot the dcd if it has something running. Could you tell me where did you put those snippet ? I will do give it a try later on.

kasjer, 2020-05-07:
Those lines were added at the end of **board_init()** for **stm32l476disco.c** just before
```c
USB_OTG_FS->GCCFG |= USB_OTG_GCCFG_VBDEN;
```

UweBonnes, 2020-05-20:
Can people have a look at my comment in #415?

UweBonnes, 2020-05-20:
Moving my comments from #415 to here:
 
STM32Cube_FW_F7_V1.16.0/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_ll_usb.c does the following:
`  /* VBUS Sensing setup */
  if (cfg.vbus_sensing_enable == 0U)
  {
    USBx_DEVICE->DCTL |= USB_OTG_DCTL_SDIS;

    /* Deactivate VBUS Sensing B */
    USBx->GCCFG &= ~USB_OTG_GCCFG_VBDEN;

    /* B-peripheral session valid override enable */
    USBx->GOTGCTL |= USB_OTG_GOTGCTL_BVALOEN;
    USBx->GOTGCTL |= USB_OTG_GOTGCTL_BVALOVAL;
  }
  else
  {
    /* Enable HW VBUS sensing */
    USBx->GCCFG |= USB_OTG_GCCFG_VBDEN;
  }
`

I only have here a STM32F746 discovery where PA9, the dedicated VBUS pin is used for USART and PC11 is available to monitor VBUS as genera GPIO. This is similar to the STM32L476disco. Adding this in board_init()

`#define DEVICE_BASE (USB_OTG_DeviceTypeDef *) (USB_OTG_FS_PERIPH_BASE + USB_OTG_DEVICE_BASE)
  USB_OTG_DeviceTypeDef * dev = DEVICE_BASE;
  dev->DCTL |= USB_OTG_DCTL_SDIS;

  /* Deactivate VBUS Sensing B */
  USB_OTG_FS->GCCFG &= ~USB_OTG_GCCFG_VBDEN;

  /* B-peripheral session valid override enable */
  USB_OTG_FS->GOTGCTL |= USB_OTG_GOTGCTL_BVALOEN;
  USB_OTG_FS->GOTGCTL |= USB_OTG_GOTGCTL_BVALOVAL;

`
makes USB enumerate with board reset or program start. Probably monitoring the VBUS GPIO in the main loop would be enough to allow hot plug.

hathach, 2020-05-22:
thanks to @kasjer  and @UweBonnes  comments, I update the stm32L476 to not use VBUS sense and force device mode, and it could enumerate just fine https://github.com/hathach/tinyusb/blob/fix-stm32l4/hw/bsp/stm32l476disco/stm32l476disco.c#L191

I verify what @kasjer observation with delay MSC and not working CDC, this is due to race condition with control transfer, probably due to the crystal-less and/or slower cpu on L4 (not sure). 

Note that the DCD report multiple 7 bytes control transfer complete comparing to the bus capture. This is a good process, I am out of bandwidth for L4, will troubleshoot it later.

[image]

kasjer, 2020-05-22:
@hathach I'm afraid it's not just race condition. Synopsys block is different from F4 boards. Interrupt for transfer complete fires for setup transfer which was not the case before. So after completing some real transfer on EP0, information about transfer size stays, and next interrupt for transfer complete (which is really for setup transfer) gets interpreted while there are not real data.
This comment is from what I could remember when I tackled it first.

hathach, 2020-05-22:
thanks @hathach for helpful information, yeah, It is just my guess, apparently it is not correct :D. Later on I will need to go through L4 USB to see if there is anything new there.

m-mcgowan, 2021-09-02:
Hi there @kasjer and @hathach!  I'm excited to see tinyusb has a definition for the `STM32L4R5nucleo` board under `hw/bsp`. Can anyone confirm that they have seen this board function correctly with the `cdc_msc` example?  Our board is very similar to this one, but I am not able to get the usb functioning. It would be great to know if this board is expected to work or not with tinyusb.  

If not, any pointers greatly appreciated before I go diving in!

m-mcgowan, 2021-09-02:
Thank you for confirming that.  I will dig in deeper today!

hathach, 2021-11-01:
while working on generalizing DWC2 driver, I find out that L4 indeed use an different revision 3.10a vs 2.81a on F407. This revision or at least on L4, both SETUP and XFRC bit is also set on DOEPINT when new set up received (while the on other mcu only SETUP bit is set). This cause dcd to register 2 events for the setup recived (1 for setup, 1 for 0 byte data). which confuse the usbd. 

Note; on F7 with revision 3.20a, only SETUP is set, so it is not sure this only happens on L4 with 3.10a, it could be caused by the actual IP configuration than the revision. Anyway, we found the root cause, fix will be push soon and hopefully add L4 to supported list.

hathach, 2021-11-01:
Update: this is indeed a bug (or feature) of the dwc2 3.10a where it  will set both OUT transfer Complete ISR first, then Setup Complete ISR afterwards in one single setup packet. confirmed in the ST hal code https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_pcd.c#L2213

I tested with L4P5, which use 3.30a and does not has the same issue. That explain why some could get it running in some variants of L4 mcu. 

Some debug logging inside the handle_rxflvl_ints()
```
EP 00, Byte Count 8, Setup Data Received
  rxflvl end
EP 00, Byte Count 0, Out Transfer Complete (ISR)
  rxflvl end
EP 00, Byte Count 0, Setup Complete (ISR)
  rxflvl end
```

hathach, 2021-11-02:
On this paritcular L476 rev 3.10a (or maybe based on actual configuration), additional transfer complete ISR is triggered in 2 event
- setup received
- control write data complete.

This has been addressed by following commit, this could be closed when generic dwc2 branch is merged. 
https://github.com/hathach/tinyusb/pull/1163/commits/aa682d730123a1191782b1a160ce25ea20580ee6#diff-1f2490fb8ceeb23801391e207e1972f09dd6a7fbbe431154211cc1896dc363a0R1003-R1023

closed 2021-11-04 by hathach

hathach, 2022-01-10:
L4 is now supported by #1163 with dwc2 driver

PR #1163 (2021-11-04): Generalize synopsys dwc2
**Describe the PR**
Generalize Synopsys DWC2 driver. New driver is at `synopsys/dwc2` For the time being, application can still pick the existing driver e.g st/synopsys/dcd_synopsys.c for stm32. Using this generalized driver is user choice until proven reliably. 

MCUs that use DWC2
- ST STM32 (F4, F7, H7 etc ..) OTG
- Espressif ESP32 S2/S3
- Giga GDVF103
- Silabs EFM32
- Broadcom bcm28/27 on RPi

Ref: linux driver for dwc https://github.com/torvalds/linux/tree/master/drivers/usb/dwc2

- implement and close #382 
- fix #1141 
- fix #126
- #750 should be re-test