hathach/tinyusb #128: Add ability for application layer to define IRQ handler

pigrew, 2019-09-10:
Could we add a dcd method for invoking the USB interrupt handler? Or would suggest how to proceed with interrupts on the ST port?

On some STM32F3, the interrupt is shared by CAN and USB, so the USB stack can't directly define the interrupt handler since both the CAN and USB handlers must be called. My board_test() module is currently failing the CI build because it isn't including the tusb drivers. Perhaps you could look at the code and suggest a fix?

It'd probably involve changing all the existing drivers, but I think the bsp should define the interrupt handler and call the dcd handler as needed. Prehaps we should add "void tud_handle_irq(uint8_t port)" which would call the appropriate "void dcd_handle_interrupt(uint8_t port)"? The port is for devices whose driver would have multiple instances (like one for FS and another for HS).

Another note is that TI-RTOS will intercept all of the interrupts, and then re-dispatch them (drivers are not allowed to directly handle the interrupts). We should think about how that would be handled, too (especially as I see someone wants to work on a MSP430 port, which would be good to be compatible with TI-RTOS).

cr1901, 2019-09-14:
Adding my own opinions before I forget...

>Another note is that TI-RTOS will intercept all of the interrupts, and then re-dispatch them (drivers are not allowed to directly handle the interrupts).

This is ultimately how Rust [real time solutions](https://github.com/japaric/cortex-m-rtfm) handle interrupts as well- setup and teardown code is required before calling the actual interrupt code because the compiler requires it.

On the other hand, it seems that FreeRTOS can happily deal with interrupts without requiring boilerplate code, so this change will make it a bit less convenient for FreeRTOS users.

rajivr, 2019-09-15:
In TockOS (non-realtime Rust OS), we do something [similar](https://github.com/tock/tock/blob/b7e7eeecd75b5479b002bf62a22b454f1ad7b02c/arch/cortex-m3/src/lib.rs#L60-L115).

I am still coming up to speed with Tinyusb code base. 

My current understanding is that interrupt enabling (`dcd_int_enable`) and disabling (`dcd_int_disable`) at NVIC is only used prevent interrupt handler from manipulating event queue, while thread mode de-queuing is happening in `tud_task`.

So if we are doing [thread mode dispatch](https://github.com/tock/tock/blob/b7e7eeecd75b5479b002bf62a22b454f1ad7b02c/chips/stm32f4xx/src/chip.rs#L44-L86) to appropriate driver, then we can safely nop `dcd_int_enable` and `dcd_int_disable`, and let the interrupt handler do its thing in thread mode.

hathach, 2019-09-19:
> In TockOS (non-realtime Rust OS), we do something [similar](https://github.com/tock/tock/blob/b7e7eeecd75b5479b002bf62a22b454f1ad7b02c/arch/cortex-m3/src/lib.rs#L60-L115).
>
> I am still coming up to [...]

yes, it is only true for No-RTOS port, for an actually RTOS, we should have an port for it, and the tud_task() will use the RTOS native queue/mutex to accessing those event queue, receive FIFO for cdc etc... 

This is an freeRTOS port, basically semaphore, mutex and queue https://github.com/hathach/tinyusb/blob/master/src/osal/osal_freertos.h

PR #325 (2020-04-13): move irqhandler to application
as discussed in #322 irq handler is moved to application, `tud_isr()` must be called to  forward the irqhandle to the stack. This PR is marked as draft and only modify the samd for reviewing purpose. If you guys agree with the changes, I will apply it to all other ports. Ports got changes so far

- [x] Microchip SAMD
- [x] Microchip SAMG
- [x] Nordic nRF5x
- [x] Nuvoton NUC120
- [x] Nuvoton NUC121/125/126
- [x] Nuvoton NUC505
- [x] NXP LPC3511
- [x] NXP LPC17/40
- [x] NXP Transdimension (18/ iMX RT)
- Sony CXD56, no need to change since it has its own interrupt i [... cut]
closed 2020-04-13 by hathach