hathach/tinyusb #129: Add USBTMC class device implementation

pigrew, 2019-09-10:
I would like a USBTMC class implementation, with USBTMC USB488 features. This request is only for a device-side driver.

(I've started working on it, but won't promise anything).

hathach, 2019-09-10:
thanks for issue, I am new to this class, I will also check to see if I could get some stubs for you. So that you don't have to read the whole stack.

pigrew, 2019-09-10:
It's actually very similar to CDC. I'll start with the CDC as a base. It mostly uses control transfers on EP0 to queue traffic on bulk. I think that interrupt endpoints may be used, too.

pigrew, 2019-09-27:
I'm closing this issue, as we've merged the patch. Yay. (PR #160 )

closed 2019-09-27 by pigrew

hathach, 2019-09-30:
Thanks for your amazing works :)