hathach/tinyusb #133: Usage of TU_ASSERT and TU_VERIFY

pigrew, 2019-09-10:
Based on other codebases, I assume that the argument of TU_VERIFY is always be executed, but the argument of TU_ASSERT is only executed if DEBUG is defined.

(i.e. the calls inside TU_ASSERT can be removed in a release build, but those in TU_VERIFY(...) are always called but their return values are only checked in a DEBUG build).

Is this how this codebase should work? Could we add some comments to the tusb_verify.h describing the intended use?

If this understanding is correct, there are a number of places where ASSERT needs to be changed to VERIFY. One example is cdc_device.c:261:

`TU_ASSERT( dcd_edpt_open(rhport, (tusb_desc_endpoint_t const *) p_desc) );`

I think that this needs to be cleaned up many places in the codebase. (And I likely have some in my STM32 port, but I have not yet looked)

hathach, 2019-09-11:
They both check for condition ALWAYS, even when DEBUG is not enabled. The macro is a bit messy, I will clean it up later. It doesn't need to be that complicated.

a `TU_ASSERT` failed is considered as fatal error, affecting the working state of the stack/device e.g failed to open an endpoint --> this cannot be ignored and will breakpoint halt when debuggger is attached, which help us to understand the cause. Or will print out message to uart/swo if CFG_TUSB_DEBUG is set and debugger is not attached (may need a bit of update deepening on IDE). 

`TU_VERIFY` is totally fine if failed, e.g checking for application input or checking busy state of endpiont etc...

pigrew, 2019-09-11:
I guess I'm really used to the .Net C# libraries where Debug.Assert's parameter is shortcutted (and not executed) when in release mode.

I consider the asserts to be code that should NEVER be triggered, so they can be removed (if desired) in a release build. Some of the MCU may have minimal flash storage, and need the assertions to be removed in order for the stack to fit, so it's convenient that debugging checks can be removed.

hathach, 2019-09-11:
Yeah, I know the builtin `assert()` is expand to nothing standard library <assert.h> . Though I still prefer to have it as it is maybe renaming it to something else such as TU_ERROR() to prevent user confusion. Though these functions are meant to be used within the libraray, application doesn't need to use it at all.

pigrew, 2019-09-12:
If documentation could be added on which to use at which time, it would be great (and I'd be happy for this report to be closed). I'd also appreciate direction on which assert to use for checks which can be removed in production code.

At the moment, I'm using TU_ASSERT() for things which only happen due to bugs in the stack or application code; these are the sort of thing I'd may want to be removed in "release" builds. I'm using TU_VERIFY() for things which can conceivably happen (e.g. if the host sends a packet with malformed data).

hathach, 2019-09-12:
@pigrew yeah sure, I should really write more documentation, but I seem always being in a middle of something lately. Will take the chance to clean it up a bit as well.

closed 2019-09-13 by pigrew