hathach/tinyusb #134: Byte endianness

pigrew, 2019-09-10:
Which functions should I be using in order to swap from USB byte order (little-endian) to the micro's byte order?

(I'm not using a big-endian micro, but it's something to think about, especially while I'm writing a class driver.)

hathach, 2019-09-11:
these one https://github.com/hathach/tinyusb/blob/master/src/common/tusb_compiler.h#L70

it is part of compiler builtin function, so far only gcc is supported, but these are easy port, probably only naming is different.

closed 2019-09-11 by hathach

pigrew, 2019-09-11:
USB is mostly little endian, so we would need to swap bytes on a big endian MCU.

Are there functions that do the swap on big endian?

hathach, 2019-09-11:
some class such as SCIS on msc use big endian https://github.com/hathach/tinyusb/blob/master/src/class/msc/msc_device.c#L250