hathach/tinyusb #135: USB MIDI Example fails to install properly on Windows

cr1901, 2019-09-11:
**Describe the bug**
On Windows, the USB MIDI Example fails to install the Audio device portion of the driver, as seen in the following images:

However, this seems to be a symptom of the bug. If I reset my STM32F4-DISC board without unplugging the USB cable, the USB MIDI device _will_ enumerate successfully when the microcontroller reboots.

**Set up (please complete the following information):**
 - OS: Windows 7
 - Board: stm32f407disco
 - Firmware Code: e.g examples/device/midi_test

**To Reproduce**
Steps to reproduce the behavior:
1. Obtain a machine running Windows 7.
2. Compile the midi_test example and load it. In my case, I used: make -C examples/device/midi_test BOARD=stm32f407disco.
3. Wait for Windows as it attempts to load drivers. This could take a few minutes the first time. **I am not sure why loading a driver only takes long the first time, even when I fully remove the device from the system each time I test.**
4. The device will fail to install, and display a notification balloon. If you click this balloon, you will get something similar to the first screenshot.
5. Reset the board.
6. The device will enumerate successfully as an audio device.

**Expected behavior**
USB device shows up as an audio device when the device is plugged in for the first time. While it's good that it works on subsequent plug-ins, I think the demo working the first time- when a computer has never seen the demo device before- is very important.

**Screenshots**
[image]
[image]
[image]

**Additional context**
I have attached a dump of Beagle USB 12 data. Windows decides to give up enumeration at between 8 and 12 seconds. At 46 seconds I reset the discovery board, and the remainder of the trace is a successful enumeration.

[usb-midi.zip](https://github.com/hathach/tinyusb/files/3598518/usb-midi.zip)

ehughes, 2019-10-23:
Has this been verified as a hardware issue or a real bug?   I am in the process of  integrating the tinyusb code (w/ midi)  into an LPC43xx project and was curious if this was still an issue.

cr1901, 2019-10-23:
@ehughes I suspect it's a real bug, but I'm waiting for a number of PRs (see PR list) that fix edge cases in USB packet handling to be resolved before I delve more deeply. We are all unfortunately busy atm it seems :(.

I _am_ able to get MIDI in to work, but not without _initial_ connection problems as portrayed above.

ehughes, 2019-11-25:
I just received an ea4357 development kit and was able build and load the project.     On this MCU,  I did not see this issue and the demo seemed to work fine.

hathach, 2019-11-25:
@ehughes are you testing with windows 7 or 10. Windows 10 seems to have better usb driver support. It can be stm32f4 specific issue, everyone (including me) are all busy now to troubleshoot this.

ehughes, 2019-11-26:
@hathach  This was under windows 10.      Everything seemed to enumerate fine.    I am going to experiment at but more with the API and will eventually test MAC and Linux

hathach, 2019-11-27:
Yes, windows 10 tends to have a better usb driver support than win7 overall