hathach/tinyusb #139: hid_composite example doesn't work with STM32 fsdev + synopsis

hathach, 2019-09-11:
**Describe the bug**
The new hid_composite example doesn't work with any of stm32 chip. The example is hid only device with only 1 input endpoint. It works on other mcu just fine 

**Set up (please complete the following information):**
 - OS: Ubuntu 18.04
 - Board: stm32f072disco, stm32f407disco
 - Firmware Code: examples/device/hid_composite

cr1901, 2019-09-11:
FWIW, I was able to get `hid_composite` working on my stm32f407 discovery board by waiting about 30 seconds before pressing the user button.

pigrew, 2019-09-12:
I build it in STM32CubeIDE... seems there is an invalid descriptor (as reported by usbtreeview):

```        ---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x00
bAlternateSetting        : 0x00
bNumEndpoints            : 0x01 (1 Endpoint)
bInterfaceClass          : 0x03 (HID - Human Interface Device)
bInterfaceSubClass       : 0x00 (None)
bInterfaceProtocol       : 0x00 (None)
iInterface               : 0x00 (No String Descriptor)
Data (HexDump)           : 09 04 00 00 01 03 00 00 00                        .........

        ------------------- HID Descriptor --------------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x21 (HID Descriptor)
bcdHID                   : 0x0111 (HID Version 1.11)
bCountryCode             : 0x00 (00 = not localized)
bNumDescriptors          : 0x01
Data (HexDump)           : 09 21 11 01 00 01 22 90 00                        .!...."..
Descriptor 1:
bDescriptorType          : 0x22 (Class=Report)
wDescriptorLength        : 0x0090 (144 bytes)
Error reading descriptor : ERROR_INVALID_PARAMETER
```

hathach, 2019-09-12:
> I built it in STM32CubeIDE... seems there is an invalid descriptor (as reported by usbtreeview):
> 
> ```
> bLength                  : 0x09 (9 bytes)
> bDescriptorType          : 0x04 (Interface Descriptor)
> bInterfaceNumber         : 0x00
> bAlternateSetting        : 0x00
> bNumEndpoints            : 0x01 (1 Endpoint)
> bInterfaceClass          : 0x03 (HID - Human Interface Device)
> bInterfaceSubClass       : 0x00 (None)
> bInterfaceProtocol       : 0x00 (None)
> iInterface               : 0x00 (No String Descriptor)
> Data (HexDump)           : 09 04 00 00 01 03 00 00 00                        .........
> 
>         ------------------- HID Descriptor --------------------
> bLength                  : 0x09 (9 bytes)
> bDescriptorType          : 0x21 (HID Descriptor)
> bcdHID                   : 0x0111 (HID Version 1.11)
> bCountryCode             : 0x00 (00 = not localized)
> bNumDescriptors          : 0x01
> Data (HexDump)           : 09 21 11 01 00 01 22 90 00                        .!...."..
> Descriptor 1:
> bDescriptorType          : 0x22 (Class=Report)
> wDescriptorLength        : 0x0090 (144 bytes)
> Error reading descriptor : ERROR_INVALID_PARAMETER
> ```

I think it is due to the fact that stm32 somehow doesn't response to host request, example run just fine on other mcu such as lpc and nrf52840. I will hook up my usb analyzer later on to troubleshoot

pigrew, 2019-09-12:
The "GET DESCRIPTOR HID Report" response is too short. Then, the computer tries to "ABORT_PIPE" and "SYNC PIPE AND CLEAR STALL". It's very likely that my stm32f0 driver doesn't handle stalls properly.

So, I think it's a combination of two bugs: The HID Report descriptor response is broken, and also our drivers don't clear stalls properly.

However, I'm about to sleep. Good luck.

dhalbert, 2019-09-12:
It may be that there is an issue in the case above, but I _always_ see the error "Error reading descriptor : ERROR_INVALID_PARAMETER" from usbtreeview, even for perfectly good HID descriptors that work just fine.

hathach, 2019-09-12:
> The "GET DESCRIPTOR HID Report" response is too short. Then, the computer tries to "ABORT_PIPE" and "SYNC PIPE AND CLEAR STALL". It's very likely that my stm32f0 driver doesn't handle stalls properly.
> 
> So, I think it's a combination of two bugs: The HID Report descriptor response is broken, and also our drivers don't clear stalls properly.
> 
> However, I'm about to sleep. Good luck.
> Here's the report response:
> 
> [image]

Good night !! I will try to fix this later on, it is systematic, so it is probably something minor. Could be my example issue as well !!!!

cr1901, 2019-09-12:
@hathach Is there a way to force a (non control EP) device stall in tinyusb for testing purposes? I can't find any obvious problems w/ my stall/clear stall impl in the Synopsys driver, but that of course doesn't mean they don't exist :).

commit 9e96c100 (2019-09-20): Merge pull request #151 from hathach/develop

fix #139, correct usbd request intepreting, Non-STD request to Interf…  … …ace must be forward to class driver
commit 340dcb81 (2019-11-04): For control transfers, compare the transmitted length against the requested length to know if a ZLP needs to happen. (fixes #139)