hathach/tinyusb #143: Lint dcd_lpc17_40

pigrew, 2019-09-11:
At dcd_lpc17_40.c, there is `TU_ASSERT((epnum % 3) == 3 && (epnum != 15))`.

This is always false since modulo 3 always returns between 0 and 2.

I'm not familiar with this device, so I'm not sure if it was intended to check for equality with 0?

hathach, 2019-09-11:
thank you very much for the PR, you are dead right, I was probably coding and sleeping at the same time back then :smile:

pigrew, 2019-09-11:
I'm playing with cppcheck; I'll push a project file for it at some point. Many of the errors are silly, but it's catching some real ones. I was really looking for a MISRA checker, but didn't find one. The TI compliers have one built-in... but yeah.

hathach, 2019-09-11:
Yeah, I used to play with those as well. MISRA 98 is really conservative which prevent all the C99 goodies and causing lots of nowadays code user C-style comment one 1 line instead of // . The next specs MISRA 2012 is better. Too bad, I bought the specs when it debuted but still haven't finished reading (almost forgot everything now). There is also CERT C/C++  coding standard as well, those are really good one. 

Though latest gcc is really smart as well, it can check many of the cpcheck rules. Maybe I need to find way to enable those :wink:

commit a37d2d64 (2019-09-16): fix #143
commit 0c70948d (2019-09-16): Merge pull request #163 from hathach/develop

fix #143
closed 2019-09-16 by hathach