hathach/tinyusb #156: Off by one error in Synopsys driver

pigrew, 2019-09-13:
Issue found by Coverity:

dcd_synopsys.c:273:

```
  // Unsupported endpoint numbers/size.
  if((desc_edpt->wMaxPacketSize.size > 64) || (epnum > EP_MAX)) {
    return false;
  }

```

I believe this is an off-by-one, and it should be `epnum >= EP_MAX`.

However, I'd suggest using TU_ASSERT with the inverse logic:

```
TU_ASSERT((desc_edpt->wMaxPacketSize.size <= 64);
TU_ASSERT(epnum < EP_MAX);
```

hathach, 2019-09-14:
I really like coverity scan now, I remembered register but struggled to get it running several years ago. This also remind me that I need to start writing some unit tests for the stack. The stack is aim to be Test-Driven Development (TDD), but heck, I never have time for such as fancy thing 😆😆

pigrew, 2019-09-14:
Is it better for me to create pull requests for these simple things, or to just do the bug report?

hathach, 2019-09-14:
It is much better if you could make an PR then fill in the information as above if it is a quick fix for you. Also you dont have to explain much, we all know if it is obvious issue. I actuall speak C better than English (non English speaker here) 😆😆

PR #158 (2019-09-14): Off-by-one error in Synopsys assertions. (and move some braces to the…
Fixes #156 

(I don't have the real hardware here, so it's untested)
closed 2019-09-14 by hathach