hathach/tinyusb #161: ep_busy_map: concurrency issues

pigrew, 2019-09-15:
The `ep_busy_map`:
* Is not declared volatile
* Is set by the `usbd_edpt_xfer()` function, which is called from interrupts, and nearly all tasks (looking at the various class examples).
* Is reset in DCD task upon receiving the DCD_EVENT_XFER_COMPLETE event.
* Can have its value read using the usbd_edpt_busy() function.
* With debugging turned off, is NOT accessed using a single-instruction cycle, so it could READ, the task be preempted and later, set the bit and write back. During the delay another task could have set a bit using `usbd_edpt_xfer()`, which would be spuriously cleared in the write-back.

I've noticed two issue with this:

* Because tud_task() never returns, it may not be guaranteed to write the value of  `ep_busy_map` back to main memory. Setting it to volatile would fix this.
* With optimization turned off 

The same issues may also apply to `ep_busy_map`, but I've not analyzed that yet.

bit clear with -O0
```
312                 _usbd_dev.ep_busy_map[ep_dir] = (uint8_t) tu_bit_clear(_usbd_dev.ep_busy_map[ep_dir], epnum);
0800289a:   ldr     r2, [pc, #148]  ; (0x8002930 <tud_task+312>)
0800289c:   adds    r2, r2, r5
0800289e:   ldrb    r0, [r2, #1]
143       static inline uint32_t tu_bit_clear(uint32_t value, uint8_t n) { return value & (~TU_BIT(n)); }
080028a0:   movs    r4, #1
080028a2:   lsls    r4, r3
312                 _usbd_dev.ep_busy_map[ep_dir] = (uint8_t) tu_bit_clear(_usbd_dev.ep_busy_map[ep_dir], epnum);
080028a4:   bics    r0, r4
080028a6:   strb    r0, [r2, #1]
```

hathach, 2019-10-01:
> EDIT: I've updated this because I misread the code initially (though the variable was accessed from within interrupts). I believe this bug only applies if using a preemptive RTOS.
> 
> The `ep_busy_map`:
> 
> * Is not declared volatile
> * Is set by the `usbd_edpt_xfer()` function, which is called from many tasks (such as the USBD core task, USBD control handler, and application tasks).
> * Is reset in USBD core task upon receiving the DCD_EVENT_XFER_COMPLETE event.
> * Can have its value read using the usbd_edpt_busy() function.
> I've noticed two issue with this:
> 
> * Because tud_task() never returns, it may not be guaranteed to write the value of  `ep_busy_map` back to main memory (perhaps given sufficient compiler optimization). Setting it to volatile would fix this (but is not a sufficient solution for MCUs with out-of-order execution).
> * `usbd_edpt_xfer()` may be called from multiple tasks "simultaneously" (such as both an application task AND the USBD task (handling a control request)). Bit setting/clearing is NOT done using a single-instruction cycle, so:
> 
> Task1	Task2
> Enter `usbd_edpt_xfer`	
> `LOAD R1,ep_busy_map`	
> Enter `usbd_edpt_xfer`
> `LOAD R2, ep_busy_map`
> `R2 |= EP1_bit
> `STORE ep_busy_map, R2`
> `R1|= EP3_bit	
> `STORE ep_busy_map, R1`	
> The same issues may also apply to `ep_busy_map`, but I've not analyzed that yet.

Thank you very much for another great analysiz. Yeah, I agree `volatile` should be added for busy and stalled map variables. There is chance that compiler could cache this value. That should probably sufficient enough for now. I am actually tentative to use `volatile` here when adding this variable, but somehow decide not include it :)

pigrew, 2019-10-01:
Yeah, volatile helps some. However, the race condition still exists. Disabling interrupts (or maybe using one byte per endpoint) should fix it. The race condition against the ISR is the more difficult case. Mutex is bad since it could be held when the interrupt is triggered.

hathach, 2019-10-01:
There is no code in isr context that access this variable. All stack code is deferred. In case of preempted rtos, the conflict is not critical at all, busy() API will return true in the next invocation

pigrew, 2019-10-01:
Yes, I was wrong when I said interrupt.

However, I think the issue I described is still a problem in a preemptive OS because setting a bit is not an atomic operation. If two tasks try to transmit simultaneously, it's possible only one busy bit will be set (when both should be set). 

Because the variable is not written from an interrupt, guarding the variable with a mutex would fix the race condition.

hathach, 2019-10-01:
Right, it is indeed an problematic issue if only 1 bit is set while there are 2 transferred. Hmm, introducing mutex is too much, I don't really like that, maybe having each endpoint has its own bool, that is 16 bytes more of sram, which is still decent enough I guess. Thank you for pointing this out, I will review usbd to see if there is better way to do this.

pigrew, 2019-10-01:
I've not yet tried out the RTOS port (is anyone using it?), but I like the concept of the stack being thread-safe, and want to use it in that mode in the future.

For the No-OS case, this bug not an issue for `ep_busy_map` now that it is volatile.

On ARM M0, FreeRTOS implements atomic instructions by disabling interrupts. I could be wrong, but my guess is that a mutex is more lightweight than disabling interrupts (or maybe it disables interrupts, anyway?). Event flags should work, but would require their implementation for the NO-OS case (so more work).

For `ep_busy_map`, perhaps create `usbd_enterCritical()` and `usbd_leaveCritical()` function which are NOP on NO-OS, but take out a mutex lock in the RTOS case, and use these to guard the global variables. However this will not work for every variable with a race condition.

Some other variables are modified from within an interrupt, such as the .connected, .configured, and .suspended flags. These are modified both in `dcd_event_handler` and in `tud_task`. Because these are part of a bit field, the entire bitfield member needs to be written only with interrupts disabled (or CAS which isn't available on ARM, though could be simulated by disabling interrupts). Unless I'm wrong (again), a mutex won't work for them.

This is becoming overly complicated. Perhaps disabling ALL interrupts while writing to the state variable is the simplest solution. Optimization can come later. However, this would also require memory barriers before reads of the globals, and code review in order to insure that the ordering of the reads doesn't matter.  Assuming that the MCU does not perform out-of-order memory operations, marking the variables as volatile is sufficient.

(I can create a patch with this solution if it's helpful, or work on some other solution that I can't think of???)

hathach, 2019-10-01:
This https://github.com/adafruit/Adafruit_nRF52_Arduino use freeRTOS. The stock example also has freeRTOS version of the cdc msc hid, but it is lacking behind now. I will update its build later on. For ep busy map, there is no need to introduce more OSAL API, I will move endpoint state to its own uint8_t byte.

pigrew, 2019-10-01:
Although less commonly used, the ep_stall_state will also need to be expanded.

commit 12609f81 (2019-10-01): fix #161 added volatile for ep_busy_map/ep_stall_map
commit d3e48da5 (2019-10-01): Merge pull request #191 from hathach/develop

fix #161 and enable -Wextra for more warnings
commit 204791b3 (2019-10-01): 2nd attempt to fix #161
commit ee63111e (2019-10-02): Merge pull request #192 from hathach/develop

2nd attempt to fix #161
closed 2019-10-02 by hathach