hathach/tinyusb #162: ST FS Dev driver: typo & EP <64 bytes

pigrew, 2019-09-16:
The ST FSDev driver has a rx/tx typo, and also doesn't properly support EP <64 bytes.

Do we have any sort of device test which we can run on a computer to verify that an driver does what it's supposed to do? I saw that the USB-IF has a (USB20-CV)[https://www.usb.org/usb2tools] tool to check if things work well for HID, MSC, video, and device. But, my laptop doesn't support it....

I also need to think about if any locking is needed in the driver, not sure yet, or if some of the globals need to be marked as volatile, etc. Can we guarantee that there won't be simultaneous requests on a single EP (both In and OUT) from multiple tasks?

Progress on patch is on a "draft pull request" (so it doesn't pollute the list in the main repository). See pigrew/tinyusb#1.

hathach, 2019-09-16:
No I don't know of device test as you mentioned. Though we can minimize these issue by writing unit tests. There are a few simple tets with ceedling (unity + cmock). I plan to add more later on when having time :)

pigrew, 2019-09-16:
I'm suggesting more for the test of the device drivers. Do you plan to emulate the USB peripherals?

hathach, 2019-09-19:
> I'm suggesting more for the test of the device drivers. Do you plan to mock/emulate the USB peripherals at the register level? Sounds complicated, though you already have the USB host written. :)

I tried once, mocking EHCI developing host stack, but let's leave that behind, it is not fun experience. The main focus of the stack is usbd, we can mock all dcd() API and make sure the stack response correctly with user descriptor and diff host behavior.

PR #165 (2019-09-20): Fsdev small packet
ST FSDEV driver updates
* Small typo fix
* Convert many of the macros to use functions. Keeping lots of stuff as `unsigned int` let the number of type conversions be reduced, reducing the final binary size by about 350 bytes (with optimization enabled). Wow. This also makes the utility functions MUCH more readable. Before the rewrite of them, I couldn't understand what they were doing.
* Implement support for EP less than 64 bytes.
* Add memory/instruction barriers after disabling interrupts
* Implement remote wakeup

These changes have been separated into three commits to m [... cut]
closed 2019-09-20 by hathach