hathach/tinyusb #164: Create usbd_open_ep() function

pigrew, 2019-09-16:
Some class drivers directly call `dcd_edpt_open`, which seems to be the the only spot where a class driver directly calls a driver function.

The usbd module does have an open EP function, but it only open pairs, not single EPs.

To this end, can a `usbd_edpt_open()` function be defined (probably just a simple wrapper), so that we can remove uses of `dcd_*` functions from class drivers? Also note that the classes are including the "device/usbd_pvt.h" header.

The changes requested:
[ ] Create `usbd_open_ep(...)`
[ ] Remove uses of the dcd functions in class drivers
[ ] Remove inclusion of the usbd_pvt header in class drivers

(For the moment, my USBTMC class is also including the pvt header file; I'm not working in implementing these changes as it is usable at the moment.)

hathach, 2019-09-16:
Yeah, it is on my plan. usbd could do even more than that. I just need to find more time 😃😃

pigrew, 2019-09-16:
Yeah, this is not high priority, just makes me feel dirty. I might do a quick, not so intellectual fix...

hathach, 2019-09-16:
It is not too intellectual, though you probably need spending more time troubleshooting bugs within the stack to be aware of edge cases :)

pigrew, 2019-09-16:
I started looking at this, and realized that they can just call `usbd_open_edpt_pair` with `count=1`.

Do they do that, or do we make a new function to open a single edpt, and make the open_pair function always open two (as its name implies}?

hathach, 2021-02-19:
implemented by #379 :smiley:

closed 2021-02-19 by hathach