hathach/tinyusb #166: USBD SET_CONFIGURATION() works only once

pigrew, 2019-09-16:
The SET_CONFIGURATION message is not handled properly if it is called to set the configuration to CFG=1 twice in a row. It also does not de-configure itself if told to go to configuration 0.

As I don't think there are plans to support multiple configurations, perhaps we should, when configuration is requested, to:  disable interrupts, tell classes to reset, and close all non-control EP?

If it's already configured, and set_configuration is called a second time, it should just return success.

hathach, 2019-09-17:
I don't think multiple configurations is supported anywhere in the world just yet. You probably couldn't find any devices and/or host driver that support it. The common way people would do it is disconnect, change the descriptor then reconnect. I don't really worry about this at all, dcd used to have edpt close() API but I removed it since there is no usage.

For `Set Config` multiple time, yes, we can add a simple if to return success. 
For `Set Config = 0` , dcd can de-configure itself when `dcd_set_config(0)` is called, e.g close all non-control endpoint. This is optional, but not really needed, since I haven't seen host issue such as request.

pigrew, 2019-09-17:
The USB compliance test software configures and configures and deconfigures many times in a row ( maybe it's just testing if the stack follows the specification....). You're probably right that nobody

I'll post a small few-line patch to "fake" the deconfigure as I outlined (mostly in order to pass the compliance test software).

hathach, 2019-09-17:
Ah, I didn't test with the compliance test software. Would you mind giving its link here, I will try to run it to test with.

pigrew, 2019-09-17:
I've finished testing, and made too many changes to the USBD core. Oops.

I'll create a PR with a link to the software.

hathach, 2019-09-19:
> I've finished testing, and made too many changes to the USBD core. Oops.
> 
> I'll create a PR with a link to the software.

That is lots of work, thank you very much.

LadislavOzobot, 2020-08-06:
Hi!
Is there any progress on this issue/PR? I have just hit this problem on a project, where we have USB CDC and vendor class. It appears linux kernel sets the configuration, and some libraries (libusb) suggest we need to do set_configuration before other operations (not doing that fails other calls, as it depends on it to set some internal structures).

We have hotfixed it with the following patch (though it is VERY ughly, it seems to work so far):
```
Index: src/device/usbd.c
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- src/device/usbd.c	(revision d05610b577f34365e3677a8811a383e4a8a00134)
+++ src/device/usbd.c	(date 1596725172765)
@@ -587,10 +587,16 @@
   return true;
 }
+static uint8_t activeCfg = -1;
 // Process Set Configure Request
 // This function parse configuration descriptor & open drivers accordingly
 static bool process_set_config(uint8_t rhport, uint8_t cfg_num)
 {
+  if(activeCfg == cfg_num){
+    return true;
+  }
+
+  activeCfg = cfg_num;
   tusb_desc_configuration_t const * desc_cfg = (tusb_desc_configuration_t const *) tud_descriptor_configuration_cb(cfg_num-1); // index is cfg_num-1
   TU_ASSERT(desc_cfg != NULL && desc_cfg->bDescriptorType == TUSB_DESC_CONFIGURATION);
 ```

hathach, 2020-08-07:
@LadislavOzobot latest commit shouldn't have issue with it, if you are not at latest, try to update it first.

hathach, 2020-10-10:
closed since it should be fixed already by now

closed 2020-10-10 by hathach