hathach/tinyusb #170: HID example: logical maximum out of bounds

pigrew, 2019-09-17:
In the composite HID example, the HID keyboard's logical maximum is out of boards.

Per the HID spec, the logical maximum is a _signed_ 1-byte value. In the code, it's setting the maximum to 0xFF and the minimum to 0x00. This is being interpreted as a minimum of 0 and a maximum of -1. The HID validator complains that 0 is not less than -1.

hathach, 2019-09-19:
interesting, I used to put value that is less than 127 there, but user reported that it needs to be 255 or so  to support non-english keyboard, the author of this PR is Japanese.

https://github.com/adafruit/Adafruit_nRF52_Arduino/commit/174ffb2df6d614207f375da9b5419a1454a7f476

I think the value should still be 255 for non-english, though we will find a way to make validator software happy hmm

pigrew, 2019-09-19:
I wonder if you can use -128 to 127?

hathach, 2019-09-19:
yeah, it could be. @ogatatsu what do you think, can we change thse to -128 to 127 and the report still work in your language ?

ogatatsu, 2019-09-19:
Hi!

When I tried to run the following code, all keys stopped working.
It seems to be interpreted as `unsigned`.
```cpp
      HID_USAGE_MIN    ( 0x80                                ), // -128
      HID_USAGE_MAX    ( 0x7F                                ), // 127
      HID_LOGICAL_MIN  ( 0x80                                ), // -128
      HID_LOGICAL_MAX  ( 0x7F                                ), // 127
```

My environment is BLE Keyboard and windows 10.

pigrew, 2019-09-19:
@ogatatsu , thanks for your reply. The issue is that the USB compliance testing fails because of these value.

I don't plan to use HID myself, so I'm happy for this issue to be closed if it needs to be.

Would you be able to post the descriptor table from other devices, so we can see if it's common to set the bounds to 0x00 and 0xFF? If it's something that everyone does, I'd suggest that we close this report, and add a comment in the source code that we know that it's against the USB HID standards, but it makes things work.

hathach, 2019-09-20:
@pigrew give me a bit of time, I will dive into it later on. Currently my hands is tied down to BLE work. The HID report descriptor is a very complicated one. We can try to see how consumer product declare this.

ogatatsu, 2019-09-20:
I captured my usb keyboard with wireshark.
Since it is long, it is pasted on gist.
https://gist.github.com/ogatatsu/95f702c64aad708827c2b26d2f562a37

It seems that 0xFF is set, but bSize is 2 bytes.
https://gist.github.com/ogatatsu/95f702c64aad708827c2b26d2f562a37#file-realforcedescriptorcapture-txt-L208

pigrew, 2019-09-20:
I've not tested it, but to check my suggestion, you should be able to change the items to be 2-byte in the table:

```
    /* 6-byte Keycodes */ \
    HID_USAGE_PAGE ( HID_USAGE_PAGE_KEYBOARD )                     ,\
      HID_USAGE_MIN    ( 0                                   )     ,\
      HID_USAGE_MAX_N  ( 255 ,2                              )     ,\
      HID_LOGICAL_MIN  ( 0                                   )     ,\
      HID_LOGICAL_MAX_N ( 255,2                             )     ,\
      HID_REPORT_COUNT ( 6                                   )     ,\
      HID_REPORT_SIZE  ( 8                                   )     ,\
      HID_INPUT        ( HID_DATA | HID_ARRAY | HID_ABSOLUTE )     ,\
```

If this works, some (static) bounds checking should be added to verify the the parameters are in the valid range. I'm not sure if everything needs to be signed, or just the logical and usage values.

ogatatsu, 2019-09-21:
> I've not tested it, but to check my suggestion, you should be able to change the items to be 2-byte in the table:
>
> ```
>     /* 6-byte Keycodes */ \
>     HID_USAGE_PAGE ( HID_USAGE_PAGE_KEYBOARD [...]

I have confirmed that my BLE Keyboard works correctly when I rewrite the firmware in the same way.

dhalbert, 2019-09-26:
To confirm this is how to represent an 8-bit unsigned value >127, here's an HID descriptor from a commercial gamepad controller (PDP one-handed controller, with one joystick and two buttons):
```
PDP one-handed controller:

0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x05,        // Usage (Game Pad)
0xA1, 0x01,        // Collection (Application)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x35, 0x00,        //   Physical Minimum (0)
0x45, 0x01,        //   Physical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x08,        //   Report Count (8)
0x05, 0x09,        //   Usage Page (Button)
0x19, 0x01,        //   Usage Minimum (0x01)
0x29, 0x08,        //   Usage Maximum (0x08)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //   Usage Page (Generic Desktop Ctrls)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x46, 0xFF, 0x00,  //   Physical Maximum (255)
0x09, 0x30,        //   Usage (X)
0x09, 0x31,        //   Usage (Y)
0x75, 0x08,        //   Report Size (8)
0x95, 0x02,        //   Report Count (2)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 45 bytes
```

hathach, 2021-05-26:
fixed by #825

closed 2021-05-26 by hathach