hathach/tinyusb #173: Add memory barrier after disabling interrupts for NO-OS?

pigrew, 2019-09-20:
I think we need a compiler memory barrier after disabling interrupts in the no-os OSAL.

My understanding is that this would only cause issues if link-time optimization is used.... In the STM32 FSDEV port, I added some extra instruction and memory barrier ASM instructions which suit this practice. These create a compiler memory barrier due to them being marked as volatile memory instructions (in the CMSIS library). 

However, I'm not sure if these instructions made it to other DCD. 

In GCC, we can use `asm volatile("": : :"memory");` to create a barrier (probably defined in the compiler settings header).

Thoughts? 'm only 80% sure this is needed (and probably nobody is using LTO). 

If you think so, feel to add the following to the compiler defines:
```
#define TU_COMPILER_BARRIER() do {asm volatile("": : :"memory");} while (0)
```
And add the call to the end of all of the dcd_disable_interrupts() functions.

(Feel free to make this change, I'm not using non-FSDEV DCD, so I'm not in a great place to test/debug it, and it's a quite simple change...)

NB: C11 has a _sync_synchronize() intrinsic which will issue the barrier ASM instructions and the compiler barrier. FreeRTOS does this already in its task switching and queue code.

hathach, 2019-09-23:
Thank you very much @pigrew , I am new to this `memory barrier`, give me a bit of time, I will try to read this up for further discussion.

pigrew, 2019-09-23:
I've also been trying to watch YouTube videos about it....

The proper solution would be to add real memory barrier instructions (which include compiler barriers). Adding the compiler barrier as I suggested is a half-solution which will work on in-order single-processor machines (I think that's the requirement...).

The proper way would be to use C11 which has the proper synchronization intrinsics, but I think this project is targetting C99, so I've been thinking about alternate solutions. 

We may want to copy/use the FreeRTOS library code as it should be well-debugged.

hathach, 2019-09-24:
ARM mcu has native instruction support 
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.faqs/ka14041.html https://developer.arm.com/docs/100941/latest/barriers

I haven't read these docs yet, maybe it can also be implemented as mcu specific and fall back to compiler if not supported. Not sure though, will spend more time to study this later on. I really want to keep this as simple and portable with less maintenance effort as possible. There isn't many user reporting this issue just yet.

pigrew, 2019-09-24:
I added the ISB and DSB instructions after disabling interrupts in the ST FSDEV driver (as that's where we are currently putting architecture-specific things).

For coding simplicity in the no-os case, adding ISA memory barriers to the disable interrupts function, and then disabling interrupts when dealing with variables accessed from interrupts is enough to be "correct" (but it would be slower). Adding ISA memory barriers without disabling interrupts would be a bit faster, and using the proper C11 synchronization primitives should be fastest.

Adding barriers in the class implementations is also useful. In my USBTMC code on STM32F0, I'm using OSAL mutexes which do create the barriers (through the disable interrupts function).

hathach, 2019-09-25:
> I think we need a compiler memory barrier after disabling interrupts in the no-os OSAL.
> 
> My understanding is that this would only cause issues if link-time optimization is used.... In the STM32 FSDEV port, I added some extra instruction and memory barrier ASM instructions which suit this practice. These create a compiler memory barrier due to them being marked as volatile memory instructions (in the CMSIS library).
> 
> However, I'm not sure if these instructions made it to other DCD.
> 
> In GCC, we can use `asm volatile("": : :"memory");` to create a barrier (probably defined in the compiler settings header).
> 
> Thoughts? I'm only 80% sure this is needed (and probably nobody is using LTO).

LTO is used more and more often i.e example of this project and circuipython repo, eventually when it is more stable, it will be as standard as other optimization option. I wouldn't want to rule out LTO at all. 

> NB: C11 has a _sync_synchronize() intrinsic which will issue the barrier ASM instructions and the compiler barrier. This is the proper way to do it. FreeRTOS does this already in its task switching and queue code. 

C11 sync is nice, but it will force a lot of project to switch/upgrade toolchain. I don't feel it is absolute must and the benefit is worth the cost. 

Right now, I don't have enough information/knowledge and time to follow this up. Let's leave it here and give me a bit more time to study it more.

pigrew, 2019-09-25:
The `__sync_synchronize()` primitive was added in GCC 4.1 (year of 2006). I propose we add a define in the compiler headers to let us use this (`#define MEMBAR() do{__sync_synchronize();}while(0)`).  This is the basic, "full memory barrier". It'll reduce performance slightly (but not as much as disabling interrupts).  We are not supporting any compilers other than GCC, so may be a reasonable approach that lets the code do what we want. If someone wants to improve performance in the future, we can slowly add other primitives.

__sync_synchronize() creates a full compiler barrier, and emits an ISA memory barrier. 

If we want, we can cause __atomic_thread_fence if the compiler supports C11. AFAIK, these instructions are identical.

The tricky engineering question here is if the barriers should be part of the DCD code, or part of the compiler code. With modern compilers, they are compiler builtins, but with old software, it is a mix of compiler and ISA-dependant ASM.

hathach, 2019-09-25:
Thanks again, your suggestion is very useful and helpful. But I really need to update/upgrade my knowledge and have a decent picture to know what to do. To be honest, membar is new to me. I haven't run into any issue with it previously.

Note: I don't rule out other compiler than gcc either :D. In general, I don't want to impose any specific limitation rather than the current C99, which allow declaring variable in the for loop(), I couldn't stand the messy iterator.

pigrew, 2019-09-25:
Alright. I also noticed a bit of stupidity on my part. The disable interrupts CMSIS function already creates a memory barrier (at least on M0 processors).

However, the enable_interrupts doesn't create a barrier. I seem to not understand the topic either. 

Enable_interrupt writes to a volatile register, but I think that the compiler is allowed to rearrange loads/stores across a volatiles (as long as its not moving another volatile), so bad things can happen.

I'll remove the extra ISB and DSB in my stm32f0 DCD code tree, but as of now it isn't hurting anything other than performance.

The TI compilers do not provide independent memory barrier instructions. They suggest using disable and enable interrupts intrinsics, which signals to the compiler that a barrier is needed.