hathach/tinyusb #179: Repeated disconnections causes the cdc_msc demo to stop functioning.

cr1901, 2019-09-23:
**Describe the bug**
While testing whether the Synopsys driver survives a USB disconnect and reset, I decided to test repeated disconnects. After enough disconnects and reconnects, the `cdc_msc` demo will stop working on my laptop; both the Mass Storage device fails to appear, and the serial port will not send or echo data. This is despite me hearing the "device connected" jingle.

**Set up (please complete the following information):**
 - OS: Windows 7
 - Board: stm32f407disco
 - Firmware Code: e.g examples/device/cdc_msc

**To Reproduce**
Steps to reproduce the behavior:
1. Compile the `cdc_msc` demo. I used the stm32f407disco board. Make sure to use [my branch](https://github.com/cr1901/tinyusb/tree/syn-unplug)- I thought the bug was related, but I'm not sure anymore.
2. Repeatedly plug and unplug the USB cable into the host/device (_this only works for boards which don't rely on the user USB cable for power_).
3. Eventually, the Mass Storage component will fail to mount in Explorer. Additionally, while the serial port will appear according to Windows, and opening it will succeed, the port will never send data (banner or otherwise).

**Expected behavior**
Repeated connections and disconnections should consistently bring back both the CDC and MSC interfaces. The MSC interface should always be responding to IN packets.

**Screenshots**
Not applicable.

**Additional context**
I have attached a zip file of the bus activity from my Beagle: [disconnect.zip](https://github.com/hathach/tinyusb/files/3640986/disconnect.zip)

The MSC interface eventually stops responding to IN packets after enough disconnects/reconnects. Eventually Windows gives up trying to talk to the device completely; this naturally also breaks the CDC portion of the demo. Unfortunately at present, I don't know enough about MSC or SCSI to debug this myself.

_I am not able to reproduce this in `hid_composite`._

pigrew, 2019-09-23:
I have been noticing this issue (or similar), too, using USBTMC. 

Does the USBD queue need to be cleared when queueing a reset event? What if transfers were queued, reset received, transfers sent, then core reset happens?

jeremyherbert, 2019-12-22:
I think I have tripped on this bug as well. I have the tinyusb device (STM32F042) connected to a powered hub, and the hub is connected to my laptop. With enough sleep/wakeup cycles on the laptop, it seems to be disconnected.

mac os seems to enumerate the device as a "Vendor specific device". I'm going to set up a second PC to try and reproduce.

[image]

jeremyherbert, 2019-12-22:
I managed to connect the debugger, and blink_interval_ms is set to BLINK_MOUNTED. So this is probably related to #209 

Some random notes:

Debugger says:
```
_usbd_dev.configured = 1
_usbd_dev.connected = 1
_usbd_dev.remote_wakeup_en = 1
_usbd_dev.remote_wakeup_support = 1
_usbd_dev.self_powered = 0
_usbd_dev.suspended = 0
```

ISTR register only has the SOF bit set, all other status flags are 0. USB interrupt is no longer triggering. NVIC shows interrupt is still enabled however.

Beagle capture went from 70kB to ~600MB in about a minute while this was all happening. Seems to be a ton of data on EP3_OUT (~140 000 handshake NACKs, see pic below ), then device re-enumerates. Device re-enumerated 7 times before getting into this stuck state.

[image]

hathach, 2019-12-24:
too many bugs :(

jeremyherbert, 2019-12-24:
I'm working on it :) But my beagle is preventing my J-Link from working, so I need to dig out an ST-Link.

xobs, 2020-04-18:
I suspect I've run into this issue on nrf.

```
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
0x0002f664 in tud_task () at ../../lib/tinyusb/src/device/usbd.c:413
413               TU_ASSERT(drv_id < USBD_CLASS_DRIVER_COUNT,);
(gdb) p drv_id
$32 = 255 '\377'
(gdb) bt
#0  0x0002f664 in tud_task () at ../../lib/tinyusb/src/device/usbd.c:413
#1  usb_background () at ../../supervisor/shared/usb/usb.c:79
#2  0x00037c68 in run_background_tasks () at background.c:62
#3  supervisor_run_background_tasks_if_tick () at ../../supervisor/shared/tick.c:90
#4  0x0004eb60 in run_code_py (safe_mode=<optimized out>) at ../../main.c:269
#5  main () at ../../main.c:467
(gdb)
```

So the `drv_id` is set to `DRVID_INVALID`, which is what it gets set to during a reset.  The reset function doesn't drain the buffer, and in fact `tu_fifo_clear()` isn't exposed in `osal_none.h`.  This seems like it fixes things by ensuring the fifo is drained:

```patch
diff --git a/src/device/usbd.c b/src/device/usbd.c
index 16670b9..fcf2b4a 100644
--- a/src/device/usbd.c
+++ b/src/device/usbd.c
@@ -328,6 +328,11 @@ static void usbd_reset(uint8_t rhport)
   {
     if ( _usbd_driver[i].reset ) _usbd_driver[i].reset( rhport );
   }
+
+  // Drain the event queue
+  dcd_event_t event;
+  while ( osal_queue_receive(_usbd_q, &event) )
+    ;
 }

 /* USB Device Driver task
```

jeremyherbert, 2020-04-18:
Do you have a dump of the USB traffic? That helped me a lot with a very similar issue, see this https://github.com/hathach/tinyusb/pull/251

xobs, 2020-04-18:
I do, but the dump just shows the host sending a reset, during which the device complains that it's processing data in the queue after the device has been reset.

[image]

Which makes sense, because the following sequence would trigger this bug:

1. Host sends a packet to the device
2. Device loads the packet into `_usbd_q`
3. Host resets the device, which calls `usbd_reset(rhport)` in an interrupt context
4. `usbd_reset()` sets the endpoint mapping to `DRVID_INVALID`
5. `tud_task()` is called to process the next event in the queue
6. `          TU_ASSERT(drv_id < USBD_CLASS_DRIVER_COUNT,);` (line 418) is hit because the `drv_id` is now `255`.

jeremyherbert, 2020-04-18:
Right, sorry. I reread your comment and it makes sense to me now. Certainly this explains the issue I was having with the STM32.

pigrew, 2020-04-18:
For STM32FSDEV, it handles reset with highest priority, then handles transfer complete messages. I don't know if that's related or not. I was thinking that a USB reset was supposed to clear the transfer complete flags, so maybe the driver itself wouldn't ever report transfers (which happen just prior to the reset).

hathach, 2020-04-18:
> For STM32FSDEV, it handles reset with highest priority, then handles transfer complete messages. I don't know if that's related or not. I was thinking that a USB reset was supposed to clear the transfer complete flags, so maybe the driver itself wouldn't ever report transfers (which happen just prior to the reset).
> 
> EDIT: actually, it doesn't prioritize reset in the handler after the first non-reset interrupt is handled. Maybe this should change, based on the resolution of this issue.

It does ring a bell to me, I think I did clear/empty the whole event queue when receiving bus reset. But somehow I decided to remove that code. Maybe it is worth to revise it.

hathach, 2020-04-18:
> I suspect I've run into this issue on nrf.
>
> ```
> (gdb) c
> Continuing.
>
> Program received signal SIGTRAP, Trace/breakpoint trap.
> 0x0002f664 in tud_task () at ../../lib/tinyusb/src/device/usbd.c:413 [...]

Ah, right, look like the slow CPU on fomu doesn't process event fast enough :smile: . Indeed the bus_reset event will immediately reset all the driver, which is right thing to do. I will take a look and maybe bring back the event queue empty up on the BUS RESET/DISCONNECTION.

xobs, 2020-04-19:
This is on an nRF52833, not on a Fomu.

hathach, 2020-04-19:
> This is on an nRF52833, not on a Fomu.

Ah right, sorry, I shouldn't read issue on midnight. I will create an patch for this.

hathach, 2020-04-19:
@xobs I have a bit of trouble to reproduce the issue with cpy, do you run any `code.py` there. I am testing with pca10056 and cpy 5.2

xobs, 2020-04-19:
I'm having trouble reproducing the problem now as well.

My installation of cpy was having many issues.  I'm on a PCA10100.  I fixed the majority of the problems, and I can't reproduce this disconnect anymore.  I wonder why it's not an issue normally.  There must be some mechanism that drains the queue if a reset hits.  Either that, or the reset pathway is not well exercised and this problem is only rarely encountered.

hathach, 2020-04-20:
> I'm having trouble reproducing the problem now as well.
>
> My installation of cpy was having many issues. I'm on a PCA10100. I fixed the majority of the problems, and I can't reproduce this disconnect [...]

I will probably pull out the fomu to test, since it run slower than other ports. It is probably easier to reproduce. I now remembered why I remove the queue clear when reset, since it isn't right. I will try to test and find a better fix for this.

proppy, 2021-07-29:
I also eems to be getting error after flashing `cdc_msc` example on the fomu,
```
make BOARD=fomu CROSS_COMPILE=riscv64-unknown-elf-
dfu-util -D _build/fomu/cdc_msc.bin
```
See below:
```
[34541.026812] usb 2-3.3.2: USB disconnect, device number 50
[34541.298728] usb 2-3.3.2: new full-speed USB device number 51 using xhci_hcd
[34541.426773] usb 2-3.3.2: device descriptor read/64, error -32
[34541.662737] usb 2-3.3.2: device descriptor read/64, error -32
[34541.898728] usb 2-3.3.2: new full-speed USB device number 52 using xhci_hcd
[34542.026771] usb 2-3.3.2: device descriptor read/64, error -32
[34542.262728] usb 2-3.3.2: device descriptor read/64, error -32
[34542.370867] usb 2-3.3-port2: attempt power cycle
[34543.882727] usb 2-3.3.2: new full-speed USB device number 53 using xhci_hcd
[34543.906624] usb 2-3.3.2: New USB device found, idVendor=1209, idProduct=5bf0, bcdDevice= 1.01
[34543.906630] usb 2-3.3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[34543.906634] usb 2-3.3.2: Product: Fomu PVT running DFU Bootloader v1.9.1
[34543.906636] usb 2-3.3.2: Manufacturer: Foosn
```

hathach, 2021-07-29:
@proppy this issue is more about stm32f4 as 1st post indicates, please open an new issue for the fomu.

proppy, 2021-07-29:
@hathach understood I thought it also applied here because of https://github.com/hathach/tinyusb/issues/179#issuecomment-615903720 and https://github.com/hathach/tinyusb/issues/179#issuecomment-616301515, will open a new issue.

hathach, 2021-08-14:
probably be fixed by #1021

duempel, 2021-08-16:
I just tested the bugfix from #1021 with a simple CDC application on a custom STM32F105 hardware.

Using the last commit 89e4586 before bugfix led into the known behavior which stopped the CDC class to work after a few reconnections.
Using the commit 48f17ef just after bugfix has always been **stable** to me. Even after 30 reconnections everything worked fine. Same positive behavior with the current master of course 😉.

hathach, 2021-08-17:
thanks @duempel for testing this out, I think we could probably close this issue for now. Should it still be an issue, we could re-open it later on.

closed 2021-08-17 by hathach