hathach/tinyusb #182: Unable to delete files via iOS 13

CollinCunningham, 2019-09-26:
Hey @hathach - We ran into an issue using CircuitPython 5.0.0-alpha.4 with iOS & iPadOS 13. Limor mentioned this would be the best way to bring it to your attention.

**Describe the bug**
Unable to delete or overwrite files on Circuit Playground Express running CircuitPython 5.0.0-alpha.4. This functionality works with CircuitPython 4.1.0.

**Set up (please complete the following information):**
 - iOS 13 & iPadOS 13.1
 - Circuit Playground Express
 - CircuitPython 5.0.0-alpha.4

**To Reproduce**
Steps to reproduce the behavior:
1. Connect CPX to iPhone or iPad via Lightning adapter or USB-C (iPad Pro)
2. Open Files app
3. Navigate to CIRCUITPY drive
4. Attempt to delete or replace code.py
(code.py may disappear, but reappears when navigating back and then reopening CIRCUITPY)

**Expected behavior**
Pasting a new version of code.py in CIRCUITPY should replace old version, prompting CPX to restart and run the new code.

**Screenshots**
https://www.dropbox.com/s/k99r1x7bjuck86r/ios13-cirpy5-a4.mov?dl=0

**Additional context**
The process we've successfully used with CircuitPython 4.1.0  & iOS 13 can be found here:
https://blog.adafruit.com/2019/09/19/program-circuitpython-devices-with-iphone-ios-13/

Let me know if I can provide any more info. Thanks!

hathach, 2019-09-26:
thanks @CollinCunningham it is as details as it can go, I will order an lighting-to-usb adapter to test with.

hathach, 2019-09-27:
@CollinCunningham I just tested CPX with current master, its seems to work with my IPX on iOS 13.1 . I could be able to remove files from CPX

CollinCunningham, 2019-10-01:
You're able to delete files from CPX using the Files app?  Strange.

hathach, 2019-10-01:
I will do more tests later on, It should be the same for iPhone 10 and 11. It is mostly the driver which is the same on both phones. Maybe I get lucky, will get you update with further testing.

CollinCunningham, 2019-10-02:
Just for reference, pasting this from our email thread

> I just loaded that new uf2 up and tested on iPad Pro (iPadOS 13.1) and was unable to delete files from CPX.
> I then updated to iPadOS 13.1.2 and had the same issue.
> 
> I’ve been loading the uf2 files onto CPX from macOS. Just in case it’s helpful - here’s a comparison between files seen on CPX running the new uf2 versus the CPX running 4.1.0 (which is working):

[image]

[...]

> Also - as an experiment, I connected CPX running new uf2(5.0.0-alpha.4-99) to a Windows 10 machine and deleted these files:
> ._boot_out.txt
> ._code.py
> .fseventsd
> .metadata_never_index
> .Trashes
> 
> I then connected the CPX to iPad Pro and was able to delete and replace files successfully.

hathach, 2019-10-02:
Another person also confirms that removing those hidden files help as well. It probably has something to do with the filesystem contents as well. 

Here is the disk image of the cpx before removing hidden files for reproduction https://www.dropbox.com/s/t3rdc0qchxsqtxb/CIRCUITPY.dmg?dl=0

hathach, 2019-10-03:
@CollinCunningham interesting note that copying all files from the dmg file to SD card. And yeah, iOS couldn't removes file on SD card. I guess it is possibly iOS bug ?!?!

CollinCunningham, 2019-10-03:
@hathach Possible the `.metadata_never_index` file is incompatible with iOS?

hathach, 2019-10-04:
@CollinCunningham after tons of trial and erorrs, I finally could determine the root causes of the issue. The present of file `.Trashes` will prevent the iOS from doing any delete operation. This file present is kind of weird, it should be a **Directory** not file. I have no idea how it appear in the CPX in the first place, but not creating it will help 

[image]

dhalbert, 2019-10-04:
That is a CircuitPython bug and I've filed an issue about that: https://github.com/adafruit/circuitpython/issues/2199. It's easy to fix.

dhalbert, 2019-10-04:
OK, it's more complicated. We deliberately made `.Trashes` a file, not a directory, because for MacOS, that prevents a "trash can" from being created on the filesystem with deleted files. It works to do that on MacOS, but apparently iOS is not acting properly. So we could report this to Apple.

hathach, 2019-10-04:
Thanks @dhalbert for clear thing up. It makes more sense now :)

hathach, 2019-10-07:
since it isn't usb issues. @CollinCunningham @dhalbert Let's me know if we could now close this issue here.

CollinCunningham, 2019-10-07:
Cool - we'll relay the bug to apple. Thanks all for the detective work!

closed 2019-10-07 by CollinCunningham