hathach/tinyusb #183: HID Gamepad stopped working between versions of TinyUSB

dhalbert, 2019-09-26:
[This was originally reported internally, but I'm writing it up here, and also now including Beagle traces for help in debugging.]

In trying support a particular gamepad in CircuitPython, we have an example of a gamepad that worked at tinyusb commit f8081536, but that is not working with tinyusb 5804e56e. This corresponds to the versions of tinyusb used by https://github.com/adafruit/circuitpython/commit/15edc9654, and the current tip of master of circuitpython.

The gamepad HID descriptor is very similar to a working one (pdp one-handed controller for microsoft XAC). The x,y coords are 0-255 and are properly a two-byte unsigned value in the descriptor. In this case it has a report id even though it is one device, but that doesn't seem to matter (I've tried with and without), and the working one has a report id even though there's only one HID device.

I tried to make the USB and HID descriptors as identical as possible in the two .uf2's below, to eliminate any issue due to the descriptors themselves.

Behavior: If the program works, you'll see four NeoPixels lit up in colors. Their intensity varies as you rotate the CPX in various directions. If the program fails, you'll see the sixth NeoPixel (numbering from 0 counterclockwise) turn green to indicate this.

1. On 18.04 Ubuntu Linux, neither .uf2 below works. This may be due to some issue with gamepad support on Linux.
2. On Windows 10, A works, but B doesn't (I have once seen B work, which makes me think it might be a timing issue).
3. On current MacOS, both A and B work.

Here are the relevant files to help reproduce this:

A. early 4.0.0 alpha CircuitPython .uf2 (works on Windows)
[gamepad-work.uf2.zip](https://github.com/hathach/tinyusb/files/3658486/gamepad-work.uf2.zip)

B. Non-working .uf2. The underlying code times out waiting for `tud_hid_ready()` in `circuitpython/shared-module/usb_hid/Device.c`. T
[gamepad-no-work.uf2.zip](https://github.com/hathach/tinyusb/files/3658485/gamepad-no-work.uf2.zip)

Add these .py files to CIRCUITPY:
[code.py.zip](https://github.com/hathach/tinyusb/files/3658539/code.py.zip)
[xac_gamepad.py.zip](https://github.com/hathach/tinyusb/files/3658540/xac_gamepad.py.zip)

Here are Beagle traces of A and B being plugged into Windows and failing or succeeding:

[Windows-A-trace.tdc.zip](https://github.com/hathach/tinyusb/files/3658721/Windows-A-trace.tdc.zip)
[Windows-B-trace.tdc.zip](https://github.com/hathach/tinyusb/files/3658722/Windows-B-trace.tdc.zip)

hathach, 2019-09-26:
thanks Dan for lots of detail :) I did write this down on my notebook, will fix it asap :)

hathach, 2019-10-01:
@dhalbert hmm I notice REPL print out the incorrect report length error, which is the reason that prevent hid to send report to host

```
code.py output:                                                                 
send report len 3                                                               
Buffer incorrect size. Should be 6 bytes.
```

The issue is raised here https://github.com/adafruit/circuitpython/blob/master/shared-module/usb_hid/Device.c#L45

A quick hid descriptor dump show the different in report size for gamepad between working and non-working uf2. 
- The working uf2 has exactly 3 bytes report size 8-bit button and X + Y 
- while the later report descriptor the gamepad is 6 bytes report with 16-bit button + X + Y + Z + Rz

I think you should probably need to upgrade the `xac_gamepad.py` to use the correct mapping for upgraded gamepad :) 

**Working UF2**
```
Usage (Gamepad),            ; Gamepad (05h, application collection)
Collection (Application),
    Report ID (5),
    Logical Minimum (0),
    Logical Maximum (1),
    Physical Minimum (0),
    Physical Maximum (1),
    Report Size (1),
    Report Count (8),
    Usage Page (Button),    ; Button (09h)
    Usage Minimum (01h),
    Usage Maximum (08h),
    Input (Variable),
    Usage Page (Desktop),   ; Generic desktop controls (01h)
    Logical Maximum (255),
    Physical Maximum (255),
    Usage (X),              ; X (30h, dynamic value)
    Usage (Y),              ; Y (31h, dynamic value)
    Report Size (8),
    Report Count (2),
    Input (Variable),
End Collection
``` 

**Non working**
```
Usage (Gamepad),                        ; Gamepad (05h, application collection)
Collection (Application),
    Report ID (4),
    Usage Page (Button),                ; Button (09h)
    Usage Minimum (01h),
    Usage Maximum (10h),
    Logical Minimum (0),
    Logical Maximum (1),
    Report Size (1),
    Report Count (16),
    Input (Variable),
    Usage Page (Desktop),               ; Generic desktop controls (01h)
    Logical Minimum (-127),
    Logical Maximum (127),
    Usage (X),                          ; X (30h, dynamic value)
    Usage (Y),                          ; Y (31h, dynamic value)
    Usage (Z),                          ; Z (32h, dynamic value)
    Usage (Rz),                         ; Rz (35h, dynamic value)
    Report Size (8),
    Report Count (4),
    Input (Variable),
End Collection
```

dhalbert, 2019-10-01:
@hathach Thank you - they are supposed to be the same descriptor, and I thought I had checked tha multiple times. I'll get back to you as soon as possible.

dhalbert, 2019-10-01:
@hathach I just downloaded the gamepad-no-work.uf2.zip from the post above, and put it on a CPX again. 

Are you using the .uf2 from above, or some other one (e.g., the tip of master?) This is a special one that has been changed to match the "working" one. It's not commited to `master` right now.

The gamepad descriptor is:
```
$ sudo usbhid-dump -d 239a
003:107:003:DESCRIPTOR         1569934732.262892
 05 01 09 05 A1 01 85 01 15 00 25 01 35 00 45 01
 75 01 95 08 05 09 19 01 29 08 81 02 05 01 26 FF
 00 46 FF 00 09 30 09 31 75 08 95 02 81 02 C0
```
which is 
```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x05,        // Usage (Game Pad)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x35, 0x00,        //   Physical Minimum (0)
0x45, 0x01,        //   Physical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x08,        //   Report Count (8)
0x05, 0x09,        //   Usage Page (Button)
0x19, 0x01,        //   Usage Minimum (0x01)
0x29, 0x08,        //   Usage Maximum (0x08)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //   Usage Page (Generic Desktop Ctrls)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x46, 0xFF, 0x00,  //   Physical Maximum (255)
0x09, 0x30,        //   Usage (X)
0x09, 0x31,        //   Usage (Y)
0x75, 0x08,        //   Report Size (8)
0x95, 0x02,        //   Report Count (2)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 47 bytes
```

dhalbert, 2019-10-01:
Also I am not getting a descriptor length mismatch, but a "USB Busy" error. So I think you are using a different "non-working" .uf2.

hathach, 2019-10-01:
@dhalbert I think upgrading tinyusb to current tip of master will fix this issue, especially after this https://github.com/hathach/tinyusb/pull/192

The issue is likely #161 issue. I make a quick hack to modify xac_gamepad.py to increase the report size to 6, and it seems to work :)

dhalbert, 2019-10-01:
Indeed, this is fixed by #161. I updated CircuitPython's version of tinyusb to the latest master, and it works.

closed 2019-10-01 by dhalbert

hathach, 2019-10-02:
Phew, I am glad it works :)