hathach/tinyusb #184: TinyUSB devices will not enumerate with 8-byte max packet size.

cr1901, 2019-09-27:
**Describe the bug**
While working on the MSP430 driver, I noticed that tinyusb would get into a state where the host would start the Status Phase of a Control Read _before_ tinyusb was done sending the packet of the data phase. This only happens during a Get Device Descriptor request immediately after a device is plugged in (or the firmware starts).

Page 39 (PDF page 67) of the USB 2.0 spec states:

>In order to determine the maximum packet size for the Default Control Pipe, the USB System Software reads the device descriptor. The host will read the first eight bytes of the device descriptor. The device always responds with at least these initial bytes in a single packet. After the host reads the initial part of the device descriptor, it is guaranteed to have read this default pipe’s wMaxPacketSize field (byte 7 of the device descriptor). It will then allow the correct size for all subsequent transactions. For all other control endpoints, the maximum data payload size is known after configuration so that the USB System Software can ensure that no data payload will be sent to the endpoint that is larger than the supported size.

I interpret this as "the first packet ever sent to the host after a device is connected connection is _negotiated_ by the host and device to be `bMaxPacketSize0`". What tinyusb does instead is schedule to send the whole device descriptor, which is always 18 bytes.

This works fine for 64 and 32-byte packet sizes, because the whole descriptor fits into a single packet. For 8 and 16-byte packet sizes, the descriptor doesn't fit into 1 packet. After the first packet is sent, the OS thinks the transaction is done and goes to the Status Phase of the control xfer- in other words, starts sending OUT ZLPs. Tinyusb doesn't know this, so it keeps trying to send the rest of the descriptor- and failing, because the hardware is waiting for an IN token the host will never send!

I made a [branch](https://github.com/cr1901/tinyusb/tree/ctl-8b) to try and accommodate 8/16 byte packets on the control pipe on STM32, but my patch does not seem to be correct. @hathach, perhaps you have an idea what's wrong because I am personally stumped right now.

**Set up (please complete the following information):**
 - OS: Windows 7
 - Board: stm32f407disco
 - Firmware Code: examples/device/cdc_msc

**To Reproduce**
Steps to reproduce the behavior:
1. Check out my [branch](https://github.com/cr1901/tinyusb/tree/ctl-8b) of tinyusb which modifies the cdc_msc demo and Synopsys driver to use only 8-byte packets on EP 0.
2. Compile the cdc_msc demo and upload it to the device.
3. The firmware should breakpoint in an xfer that is marked as complete after only sending 32 out of 64 bytes (`xfer->queued_size` will say 24; this is due to how `transmit_packet` works, but 32 bytes were queued).

**Expected behavior**
Upon device insertion/firmware init, tinyusb should expect the OS to terminate a Get Device Descriptor xfer after sending only one packet, _regardless_ of packet size. This only applies to the first Get Device Descriptor request.

Also, my attempted fix for the behavior described above should not break the Synopsys driver :P.

**Screenshots**
TODO

**Additional context**
TODO

pigrew, 2019-09-27:
I don't have this setup to test at the moment, but could you try my ZLP pull request #149 (perhaps merge it into your branch... not sure) ? I don't know if it'll help, but there is a good chance. On the other hand, I wrote the code when I didn't know at all what I was doing with respect to USB.

My interpretation is that the host must request 8 bytes (wLength=8) before it knows that the device supports longer packets. The device must return only the first 8 bytes in the data packet.

If the max packet size is 8, then the device must send a ZLP. after the 8-byte packet. If the max packet size is >8, then the data phase is complete.

The host will know if it should expect a ZLP or if it should send a status byte based on the maxpacket size contained within the first 8 bytes.

pigrew, 2019-09-27:
I found a silly typo, will create PR soon.

The USB descriptors use `CFG_TUD_ENDOINT0_SIZE`, but the USBD core uses `CFG_TUD_ENDPOINT0_SIZE`... not if this will fix everything or not.

cr1901, 2019-09-27:
>not if this will fix everything or not.

I'll cherry pick this into [my branch](https://github.com/cr1901/tinyusb/tree/ctl-8b) and see what happens. Without the extra patches in that branch, I don't think your PR will be sufficient to fix the issue.

pigrew, 2019-09-27:
@cr1901 With the new PR #186 , I am able to use USBTMC with 8 byte EP0 on STM32F0.... even without the ZLP patch.

Anyway, I have to go eat now, so I can't do much more testing.

cr1901, 2019-09-28:
@pigrew Your patch works for stm32f4disco as well. I have no idea how/why, but it works. And I'm not sure I want to know right now :P.

I believe this paragraph still applies:

> For 8 and 16-byte packet sizes, the descriptor doesn't fit into 1 packet. After the first packet is sent, the OS thinks the transaction is done and goes to the Status Phase of the control xfer- in [...]

However, with your patch, tinyusb recovers after reset; without this patch, tinyusb remains in a mismatched state w/ the host (i.e. IN EP 0 is never given a new buffer).

I've attached a screenshot and [trace](https://github.com/hathach/tinyusb/files/3664904/ep0-8b.zip) showing this behavior.

[image]

cr1901, 2019-09-28:
FWIW, my [changes](https://github.com/cr1901/tinyusb/commit/aef7661ea8a321de28ed6dd09e2bad82c41a927b#diff-e546d6e4c4e6a5c18b285fbe07514f89) to usbd.c fix the OUT-DATA-NAKs. I'm considering removing the `#if CFG_TUD_ENDPOINT0_SIZE < 32` to make things cleaner at the cost of a few extra insns on each Get Descriptor request regardless of max packet size.

pigrew, 2019-09-28:
I need to better read the spec. Based on your (and my) logs, the host (Win10 in my case) requests wLength=0x40 (64 bytes) of the device descriptor.

Since there should be more data left in `EP0/IN`, It seems weird to me that the host switches to `EP0/OUT`.

After a CONTROL/IN (like this get descriptor) The STM32F0 manual suggests setting EP0/OUT to STALL prior to the last data packet. If this were the case, then the NAK should be a STALL since the last data packet has not yet been sent, and the host could recover from this.

We should double-checked based on "commercial devices" (like FTDI or similar) if a STALL on EP0/OUT is expected after the initial read in order to fix the NAK/reset issue.  However, I'm not sure if we would be able to find such a device since it would need its max packet size to be 8, which probably nobody does.

Since the DCD is given multiple packets of data at once, the DCD drivers would need to handle setting the stall/unstall on the control endpoints at the proper times.  (we can't push the logic to the USBD core since the core isn't notified of every packet transmitted). 

This weekend, I will update PR #181 with the stall logic on EP0, and post a log of what happens.

cr1901, 2019-09-28:
>Since there should be more data left in EP0/IN, It seems weird to me that the host switches to EP0/OUT.

The host doesn't know the max packet size. So the host negotiates a transfer in the Setup phase for the maximum `bMaxPacketSize` that a device can possibly support (64). A USB xfer terminates if the device sends a short packet, which is the case for `bMaxPacketSize` of 8, 16, and 32, or _exactly_ the amount of data negotiated between host and device, which is 64.

So it makes sense to me that the Data Phase terminates after a single packet.

pigrew, 2019-09-28:
> A USB xfer terminates if the device sends a short packet, which is the case for `bMaxPacketSize` of 8, 16, and 32, or _exactly_ the amount of data negotiated between host and device, which is 64.

For 8 or 16 packet sizes, with a 64 byte transfer requested (Win10 with a 8-byte packet device), the host will think it's a short packet and send it to its stack after receiving the 8-byte data1. The stack could then analyze the data in the first 8 bytes, set the packet size properly, and then add requests for more of the data until an actual short packet is received. However, I don't know if the USB host driver on Windows would support this sort of operation. My guess is that the Windows device driver automatically sends a status packet after receiving what it believes to be a short packet.

The OUT-DATA-NAK seems to be a status packet to assuming that the device had actually finished the transaction with what the host believed to be a short packet. The device has no clue that the host things it's done. Two ways to exit this state would be a STALL response to the ZLP (since it's too early), or the host just resetting the device. I thought that sending a new SETUP would reset it, but perhaps the Windows USB drivers couldn't do that, either.

(My mouse isn't a good example to look at since it's low-speed which have 8-byte packets, anyway... Need a full speed 8-byte packet device...)

pigrew, 2019-09-28:
Here's a log of a RS232 ->USB converter which is full-speed with 8-byte EP0 packets. It uses TUSB3410 chip. 

1. Windows requests the device descriptor (packet 33)
2. 8-byte IN reply (packets 68-70)
3. Host sends 4 CONTROL OUT DATA (all NAKed by device)
4. Windows sends reset.

I don't know why in your log it had so many ZLP and mine only has 4. Perhaps it's a Windows 7 vs Windows 10 thing.

(The log is from DSView... DreamSourceLabs)
[rs232_converter_8byte_packet.zip](https://github.com/hathach/tinyusb/files/3665082/rs232_converter_8byte_packet.zip)

cr1901, 2019-09-28:
>My guess is that the Windows device driver automatically sends a status packet after receiving what it believes to be a short packet.

Yes, that is what I was trying to get at in my previous message :P.

>I don't know why in your log it had so many ZLP and mine only has 4.

Hmmm, interesting. And I'm not sure :(...

pigrew, 2019-09-28:
I've tried sending STALL in response to the "unexpected" STATUS OUT packet, and Windows 7 really doesn't like it.

If Windows 7 sees a stall on the STATUS OUT, it will abort the enumeration (after retrying three times). So, because of Win7 compatibility, we cannot STALL the spurious status packet:

[stall_status.zip](https://github.com/hathach/tinyusb/files/3665561/stall_status.zip)

[I found a link describing the Windows enumeration logic](https://techcommunity.microsoft.com/t5/Microsoft-USB-Blog/How-does-USB-stack-enumerate-a-device/ba-p/270685). It says because of bugs with older USB devices, it ALWAYS sends a reset after the initial first device descriptor request.

So, it looks like your approach to ACK the "unexpected" (what seems to be non-USB compliant) is the only way to work with the Windows 7 host stack quiet down.

Open question: If the status is ACKed, will enumeration happen faster on Win7? Win10?

So, what heuristic to use to queue the ACK on EP0 OUT? And should it be in the device driver or USBD? And except this initial request, should we stall in the opposite direction?

I would like to leave open the option to stall the opposite direction endpoint to make error recovery faster, and with the stack architecture, that logic needs to be in the DCD ISR. So, let me propose a design:

1) For responses, USBD submits entire response (more than a packet) to the DCD for transmitting. Once the transmit complete is routed to USBD, it will queue a ZLP if needed (based on short packet logic).
2) USBD_control keeps track of if the EP gone through the enumeration process, perhaps a `presubmit_status_ack` flag. This bit is SET when a USBD reset happens, and cleared when a GET_CONFIGURATION_DESCRIPTOR SETUP packet is received.
3) When transmitting a control response (EP 0x80), after transmitting the CTRL IN data to the DCD, a  CTRL OUT status RX transaction is immediately started.
4) In the USBD_ctrl xfer complete handler, the status packet is only sent if `presubmit_status_ack` is false.

On my STM32F0 board, I cannot tell the difference between a DISCONNECT and a RESET (VBUS is not connected to the STM32F0), so I don't want the bit to survive a RESET. Also, the user can request that Windows reset a device without removing vbus, so the MCU would not know that a new enumeration is happening (or maybe it'll remember the EP0 size during the requested reset, not sure).

I suggest resetting the flag on GET_CONFIGURATION_DESCRIPTOR so that no code is added for most control transfers and a configuration descriptor will be read soon after the device descriptor. Perhaps there is a transmit error in the TUSB_DESC_DEVICE transmit, so we don't want to immediately clear the bit at that point.

(I'll implement this later today. I don't know how well it'll work)

cr1901, 2019-09-28:
>It looks like you are working on the implementation?

No, I'm not working on an implementation. I have found this patch to be sufficient for my needs:

```diff
diff --git a/src/device/usbd.c b/src/device/usbd.c
index a6eda79d..2f9e7406 100644
--- a/src/device/usbd.c
+++ b/src/device/usbd.c
@@ -1,4 +1,4 @@
-/* 
+/*
  * The MIT License (MIT)
  *
  * Copyright (c) 2019 Ha Thach (tinyusb.org)
@@ -46,6 +46,7 @@ typedef struct {
     volatile uint8_t connected    : 1;
     volatile uint8_t configured   : 1;
     volatile uint8_t suspended    : 1;
+    volatile uint8_t ep0_size_known : 1;
 
     uint8_t remote_wakeup_en      : 1; // enable/disable by host
     uint8_t remote_wakeup_support : 1; // configuration descriptor's attribute
@@ -211,6 +212,8 @@ bool usbd_init (void)
 
 static void usbd_reset(uint8_t rhport)
 {
+  uint8_t ep0_size_known = _usbd_dev.ep0_size_known;
+
   tu_varclr(&_usbd_dev);
 
   memset(_usbd_dev.itf2drv, 0xff, sizeof(_usbd_dev.itf2drv)); // invalid mapping
@@ -222,6 +225,12 @@ static void usbd_reset(uint8_t rhport)
   {
     if ( usbd_class_drivers[i].reset ) usbd_class_drivers[i].reset( rhport );
   }
+
+  // The endpoint size is known by the host as soon as the first
+  // "Get Device Descriptor" request succeeds. This knowledge survives a reset
+  // (but not a device disconnect).
+
+  _usbd_dev.ep0_size_known = ep0_size_known & 0x01;
 }
 
 /* USB Device Driver task
@@ -595,6 +604,12 @@ static bool process_get_descriptor(uint8_t rhport, tusb_control_request_t const
   switch(desc_type)
   {
     case TUSB_DESC_DEVICE:
+      if(!_usbd_dev.ep0_size_known)
+      {
+        bool rc = tud_control_xfer(rhport, p_request, (void*) tud_descriptor_device_cb(), CFG_TUD_ENDPOINT0_SIZE);
+        _usbd_dev.ep0_size_known = 1;
+        return rc;
+      }
       return tud_control_xfer(rhport, p_request, (void*) tud_descriptor_device_cb(), sizeof(tusb_desc_device_t));
     break;
 
@@ -665,6 +680,7 @@ void dcd_event_handler(dcd_event_t const * event, bool in_isr)
       _usbd_dev.connected = 0;
       _usbd_dev.configured = 0;
       _usbd_dev.suspended = 0;
+      _usbd_dev.ep0_size_known = 0;
       osal_queue_send(_usbd_q, event, in_isr);
     break;
```

Your proposal probably works fine, but I think it's adds a lot of complexity. I would rather you try breaking my patch before we resort to fun stuff like "unstall has to be in the ISR".

I still interpret the spec as "After connecting a device _and/or_ power on, the host and device implicitly negotiate that the first packet sent should be `bMaxPacketSize`". I based this off the following:

>9.1.2 Bus Enumeration
...
>4. The hub performs the required reset processing for that port (see Section 11.5.1.5). When the reset signal is released, the port has been enabled. The USB device is now in the Default state...
>5. The host assigns a unique address to the USB device, moving the device to the Address state.
>6. Before the USB device receives a unique address, its Default Control Pipe is still accessible via the default address. The host reads the device descriptor to determine what actual maximum data payload size this USB device’s default pipe can use.

Additionally, `GET_DESCRIPTOR` is one of the only device requests that work in the Default State (though the spec doesn't specify which descriptors are valid in the Default State, only a device and configuration descriptor have to be supported).

pigrew, 2019-09-29:
I started to write my solution (pushed to https://github.com/pigrew/tinyusb/commits/ctrl_test_2 , which also contains my ZLP patch).  I don't like this particular patch much, but designing a better patch is proving difficult. I quickly tested that it does function with the hid_composite demo, enumerating almost instantly on Win10, and within 1.5 seconds on Win7.

My ZLP patch was needed in order for the HID control requests to be successful (my guess is that one of the HID descriptors was a multiple of 8 bytes long).

Pre-transmitting the status on GET_DESCRIPTOR while in the default state may not be strictly standards-compliant, but the only harm I can imagine is perhaps making the EP stuck after a transfer error. The real issue with my patch is that the extra data packets will still be in the EP0 transmit FIFO, and I don't know how to clear the transmit FIFO with the current DCD interface.

I'd appreciate @hathach's input on if & how this issue should be resolved. Do we leave the behavior as-is and live with slow enumeration? Or do we add a hack of some sort to make enumeration faster?

@cr1901, the issues I see with your solution include:

1) It queues only the first packet, not the entire descriptor. Standards-compliant hosts may want to continue reading the rest of the descriptor after the first 8-byte packet, instead of resetting everything or reading on a second transfer. Does Linux perform the reset after reading the first packet? Other stacks?
2) Some USB devices might not be able to tell the difference between a reset and a disconnect (only difference is if vbus disappears). Using _disconnect to reset the flag seems fragile, unless we say all DCD must send _disconnect when they receive a reset if they are not sure?
3) size_known is set upon queuing the packet to be transmitted, not the success of the data or status transfer. If the request is corrupted (due to electrical interference), and the host performs a second query without a reset, then the full descriptor would be sent (re-creating the long delay).

cr1901, 2019-09-29:
> It queues only the first packet, not the entire descriptor. Standards-compliant hosts may want to continue reading the rest of the descriptor after the first 8-byte packet, instead of resetting [...]

I don't know.

>Some USB devices might not be able to tell the difference between a reset and a disconnect (only difference is if vbus disappears). Using _disconnect to reset the flag seems fragile, unless we say all DCD must send _disconnect when they receive a reset if they are not sure?

It's possible, I haven't seen such devices in practice, but point taken.

>size_known is set upon queuing the packet to be transmitted, not the success of the data or status transfer. If the request is corrupted (due to electrical interference), and the host performs a second query without a reset, then the full descriptor would be sent (re-creating the long delay).

Indeed this is a problem. I can move the  `size_known` bit set to after a successful data phase.

One thing both my synopsys and msp430 driver assume on entry to `dcd_edpt_xfer` is that the endpoint is not actively sending or receiving upon entry. This varies on hardware, but typically you set a `NAK` bit and possibly another bit that prevents internal counters from keeping track of sent bytes/bits. 

If we accept that for this edge case the host and device can have a mismatch on the actual xfer size and use your/libopenpcm3's solution, I think there should be a way to "cancel" an xfer (`dcd_xfer_cancel`) to avoid garbage being sent/received by the host and accidentally triggering a spurious "packet sent/recv'd" interrupt. Note spurious interrupts should of course never happen, and in practice preparing a new packet even when the endpoint is actively sending/receiving doesn't break things. But it's good to actively squash potential race conditions.

pigrew, 2019-09-30:
Reading the USB spec paragraph you have oft quoted, "The host will read the first eight bytes of the device descriptor." Based on that, what Windows does is not compliant. Following the standard, wLength should be set to 8 bytes, but it is actually setting it to 64 bytes. We are trying to put a bandaid around the bad behavior of Windows. Though, it's certainly possible that other operating systems. Reading the ST Micro USB host middleware, it sets wLength to be 8 on the first transaction.

>>Some USB devices might not be able to tell the difference between a reset and a disconnect (only difference is if vbus disappears). Using _disconnect to reset the flag seems fragile, unless we say all DCD must send _disconnect when they receive a reset if they are not sure?

> It's possible, I haven't seen such devices in practice, but point taken.

The STM32F103 blue pill is one such board. It doesn't have a connection from Vbus to an IO pin. The ST FSDEV peripherial does not have any related VBUS pin, either. One has to use GPIO to sense if Vbus is present, and the driver I wrote never triggers a disconnect event (as I don't want to mess up the user's GPIO configuration). My experience is that OTG-capable peripherals will support disconnect detection, but device-only will not.

cr1901, 2019-10-03:
@pigrew I had to merge #149 into the msp430 fork. The HID composite demo won't work without it for various reasons.

As I just found out, my proposed solution here interacts badly with your PR #149; as soon as the initial packet is sent, your ZLP-aware code (correctly!) tries to send a ZLP, which is unexpected from the host POV.

In the case of the initial setup packet, no ZLP should be sent. As long as a host is well-behaved, however, I imagine that a tinyusb device will overwrite the ZLP transfer with a new packet (based on the next SETUP packet) by the time another IN packet arrives.

pigrew, 2019-10-03:
Yes, with a well-manored host, no ZLP should be sent. However, the goal is to appease the MS Windows and make it enumerate faster. The only way to be both (nearly) standards-compliant and appease Windows is to preemptively send the ZLP (unless there is a solution I'm missing).

It's good to hear that #149 is being helpful in your MSP430 efforts.

closed 2020-03-24 by hathach

benabramz1, 2022-09-03:
Hello friends
I have such a problem with the keyboard.
I want to connect as a keyboard to a device that only supports USB1.1.
The microcontroller I use is ESP32-S2 with tinyusb. I have changed CFG_TUD_ENDPOINT0_SIZE in tusbconfig to 8, but No connection was made.

PR #194 (2020-03-24): [WIP] MSP430 Support
Do not merge this yet. 

I got `cdc_msc` working properly yesterday on the msp-exp430f5529 launchpad. However, there are a few issues I need help fixing in tinyusb before I feel comfortable having this merged:

- [x] Fix #184. I have a [stopgap solution](https://github.com/hathach/tinyusb/issues/184#issuecomment-536221082) I've been using for the time being. However, my patch breaks when combined with #149, as I described below.

When testing the `hid_composite` demo, I was getting a number of STALLs that appear to have been hardware generated (traces/pics upon request). I [... cut]