hathach/tinyusb #188: Where is EP0 unstalled?

pigrew, 2019-09-30:
Assume that a bad control request was made, and EP0 is stalled (src/usbd_control.c:146).

Where in the code should the EP be unstalled?

My working assumption is that it should be unstalled very soon after a SETUP packet is received?

hathach, 2019-10-01:
There is none, EP0 doesnt need to be unstalled. USB says that whenever the new setup packet is received the hardware should clear the stalled bit of control ep0. 

[image]

pigrew, 2019-10-01:
Thanks. I was seeing some weird behavior on ST FSDEV the other day. I'll need to debug it further. Worst case, I'll have to manually unstall it in the ISR when it receives a SETUP packet. I'll test it in the next few days.

So, I'm I'm understanding correctly, Best case is that it is unstalled by hardware. Worst case the DCD module should unstall it upon receiving a SETUP.

hathach, 2019-10-01:
> Thanks. I was seeing some weird behavior on ST FSDEV the other day. I'll need to debug it further. Worst case, I'll have to manually unstall it in the ISR when it receives a SETUP packet. I'll test it in the next few days to see if the hardware does perform the unstall itself.
> 
> So, If I'm understanding correctly, best case is that it is unstalled by hardware. Worst case the DCD module should unstall it upon receiving a SETUP.

It is indeed DCD's reposibility to unstall both EP0 IN and OUT when receiving a new setup packet. Some IP controller do that automatically by the hardware, other require dcd to clear a bit. In general tinyUSB stack won't attempt to unstall control endpoint. It is DCD work :)

pigrew, 2019-10-01:
Thanks, this was the main purpose of #181 ... I'll double check that it's correct, and probably leave the "expect 0-length packet" bit always disabled for the time being (let the stack handle that).

The FSDEV has a few error-checking features that seem optional, and would interact badly with how Windows does device enumeration.

closed 2019-10-01 by pigrew