hathach/tinyusb #197: Improper endpoint count causes failure on F412 and similar boards

hierophect, 2019-10-07:
**Describe the bug**
On the F412 or any other ST board with 6 FS endpoints which is using all of them, the following assert fails: 

```
dcd_edpt_open (rhport=0 '\000', desc_edpt=0x8068862 <usb_desc_cfg+194>) at ../../lib/tinyusb/src/portable/st/synopsys/dcd_synopsys.c:273
273	  TU_ASSERT(epnum < EP_MAX);
```

A debugger reveals that epnum is evaluating to 6, potentially because it includes endpoint 0, or possibly due to a bug. EP_MAX is also 6, as it's based off the raw endpoint macro for the chip, and thus the comparison fails. Only 5 endpoints, excluding control, are used on the F412 in this case, have worked successfully in every version prior to https://github.com/adafruit/circuitpython/pull/2185

**To Reproduce/Fix**
Use an F412 with 5 active USB endpoints.

hathach, 2019-10-07:
@hierophect It is actually a bug that allow the endpoint 6 to be opened and is fixed recently by https://github.com/hathach/tinyusb/pull/158
STM32f412 only has 6 pairs of endpoint including endpoint 0, which mean we can only go up to endpoint number 5 at max. which usb classes that is used by the cpython in this case cdc + msc + hid + ?

[image]

hierophect, 2019-10-07:
Currently the Circuitpython USB descriptor only outlines 5 endpoint pairs; cdc_iad, cdc_interfaces, msc_interfaces, hid_interfaces, and audio_control_interface. All share their endpoint address except certain cases for the SAMD21, which are of course excluded on the STM32. So I don't think we should be hitting this ceiling. 

Does epnum specifically exclude the control endpoint? My debugger shows it at 6, which the appropriate number of endpoints if control is included...

Paging @dhalbert for commentary

hathach, 2019-10-07:
@hierophect EP_MAX=6 is including endpoint0, so we can only have up to 5 pairs of endpoints from 1 to 5 at max. So the code is correct, if your debugger show epnum is 6, then it probably overflow the number of supported endpoint on stm32f412

hierophect, 2019-10-07:
Did some internal investigation, was due to extra endpoints in the MIDI descriptor. Thanks for helping clarify, @hathach, and for fixing the excess endpoints bug.

closed 2019-10-07 by hierophect