hathach/tinyusb #198: Port STM32F4 Highspeed core

hathach, 2019-10-08:
**Porting layer**
 - [x] Device Controller Driver (DCD)
 - [ ] Host Controller Driver (HCD)
 - [ ] Board Supported Package (BSP)
 - [ ] OS Abstraction Layer (OSAL)

**Description**
It is more likely a test than port, probably adjusting base address etc to get the OTG HS core work with the stack. Need to find an dev kit that support OTG HS connection.

hathach, 2019-10-09:
2 STM32F7 board with OTG HS connection are on order 
- STM32F723E Discovery with builtin HS PHY
- STM32F769I Discovery require external PHY for OTG HS

pigrew, 2019-11-04:
The new $10 st-link mini v3 have a STM32F7 and would be a nice target platform, too.

hathach, 2019-11-04:
> The new US$10 st-link mini v3 have a STM32F7 and would be a nice target platform, too. The board uses the integrated PHY and has USB ESD protection. One would need to solder on a few pins for JTAG to the board, though.

That is a very good tips, though I already bought those boards. It is still nice to get up with a known good board, running chip vendor stock example to verify hw work etc.. and I don't have much love with soldering as well :). I am sure others will find it useful.

UweBonnes, 2020-05-15:
Any news on STM32 USB-HS? Modding the StlinkV3 for blackmagic would be a nice target.

UweBonnes, 2020-06-20:
Even having the StlinkV3 HS USB only work as FS would be a big plus.

hathach, 2020-06-20:
@UweBonnes I don't have stlink V3 to test with ( I wished you remind me earlier since I just placed an order from Digikey last week for esp32s2). Good news is the stm32 hs is working well with stm32h743eval if you happen to have the board https://github.com/hathach/tinyusb/tree/add-stm-hs . It still needs a bit more testing though. 

Meanwhile if you have the board would you mind adding an BSP for it,  It doesn't need to have running USB just yet, just have the `board_test` example is a good start ( it ensure we correct set clock and have gpio running). The example will blink at 1000 ms normally and blink at 250 ms when button is pressed. Afterwards, I could try to add some USB specific code and we can see if  that could get USB running on stlinkv3  

PS: I am currently tied with esp32s2 work,thing will be rather slow.

PR #447 (2020-07-08): Add support for STM32 OTG HS core
**Describe the PR**
This PR add support for stm32 otg hs core, fix #198  board with OTG HS connection for testing.
 
- [x] STM32F723E Discovery with builtin HS PHY
- [x] STM32F769I Discovery require external PHY for OTG HS
- [x] STM32H743Eval with external PHY for OTG HS

**Additional context**
~~#439 update this add-stm-hs branch to support more boards and core with internal HS PHY.  It is best to merge 439 first to update this PR before merging.~~ merged already

@UweBonnes @cr1901 @duempel
closed 2020-07-08 by hathach