hathach/tinyusb #201: how is IRQ priority level for the chosen?

dhalbert, 2019-10-21:
@hierophect and I are trying to figure out where the IRQ priority level for the USB interrupt handlers is set, and I'm not seeing that. Could you describe where that is? He's having trouble on the STM32F4 with USB getting starved out when another peripheral's interrupts are set at level 2. There may be a bug with code for that peripheral causing the starving, but we don't really know.

In https://github.com/adafruit/circuitpython/blob/84c0d6cdf8c0bff1b61ce2fa6edcd2b4a65b2556/ports/atmel-samd/tick.c#L84 we set the USB IRQ's to prio 1 on SAMD21 and SAMD51. But this code was written long ago before CPy started using TinyUSB, so I don't even know if it works anymore or if TinyUsb is overriding these choices.

hierophect, 2019-10-21:
Specifically, the only call to I see to NVIC in the stm32 is this mention in dcd_synopsys.c:

```
/Users/lucian/Documents/Firmware_Projects/circuitpython/lib/tinyusb/src/portable/st/synopsys/dcd_synopsys.c:
  223  {
  224    (void) rhport;
  225:   NVIC_EnableIRQ(OTG_FS_IRQn);
  226  }
  227  
  ...
  229  {
  230    (void) rhport;
  231:   NVIC_DisableIRQ(OTG_FS_IRQn);
  232  }
```

This simply enables the interrupt, but does not specify priority. I assume this will end up setting it to the default priority of 0. Should this have a default priority that should then be overridden in tick.c?

hathach, 2019-10-22:
@dhalbert tinyusb won't try to change the IRQ priority at all since it varies from mcu to mcu from application to application. This should be done by application, best to be somewhere near the `tusb_init()`

hathach, 2019-10-25:
@dhalbert @hierophect Let's me know if you still have question regarding this.

closed 2019-10-29 by hathach