hathach/tinyusb #205: Class code driver conflict application specific 0xfe between USBTMC and Runtime DFU

hathach, 2019-10-28:


pigrew, 2019-10-28:
Yep, I remember us talking about this a while ago.

I believe the solution is to add subclass definitions to the callback table. (Or to create a new functions something like `class_suppors_intf(class,subclass)` which would be iterated through, which would allow for a class to support multiple subclasses...)

We also talked about moving the callback tables to the class header files instead of in `usbd.c`. Perhaps that could be changed at the same time as supporting subclass codes.

hathach, 2019-10-28:
@pigrew yeah, I just become a real issue after Runtime DFU PR is merged #202 which also use Application Specific class 0xFE. I am on multiple fronts now, will definitely refactor class driver to fix this issue when possible :)

PR #342 (2020-04-15): add subclass and protocol for matching driver id
fix #205 , this help to resolve conflict of currently class driver. Tested with msc_cdc, hid_composite, and net_lwip_webserver with nrf52840. All seems to work
closed 2020-04-15 by hathach