hathach/tinyusb #226: Adding contributor information to readme or seperate contributors file

PTS93, 2019-11-24:
Given that TinyUSB is getting a lot of traction and a lot more ports are coming in. Maybe its a good idea to name all the people who did work for specific ports?
The readme or a contributors file sound like a good idea. I think that would be rather valuable for the future if you want to contact specific people for certain ports.

hathach, 2019-11-25:
@PTS93 Yeah, you are absolutely right. I should have done this earlier. I am sure the list will grow lengthy in the future, so I think it is best to put people with their work on separated files, and have thanks note & link to that from readme. Thank you very much for reminding :)

hathach, 2019-11-25:
@PTS93 just go through the overall development and add this in https://github.com/hathach/tinyusb/pull/227 . Let's me know if it looks OK :)

PTS93, 2019-11-25:
Looks great!
(Very kind of you to include me, I didn't expect that for just an example)

closed 2019-11-25 by PTS93

hathach, 2019-11-25:
@PTS93 I tried my best to include everyone involved :)