hathach/tinyusb #229: TU_ASSERT when host stalls an MSC endpoint

xobs, 2019-11-26:
Still trying to track this one down.

The device is presenting a USB disk drive that never goes ready.  Linux is constantly sending `Test Unit Ready [0]` and `Request Sense [0]` requests, all of which get responded to correctly.

Eventually, Linux sends this request:

```
radix: hexadecimal
55 53 42 43
DD 01 00 00
08 00 00 00
80 00 0A 25
00 00 00 00
00 00 00 00
00 00 00 00
00 00 00
```

This gets responded to with a `STALL`, which the host responds with by halting the endpoint.

[image]

This triggers an exception, since the request is not a class type:

```
0x20091544 in mscd_control_request (rhport=0 '\000', p_request=0x1001ff4c) at ../../lib/tinyusb/src/class/msc/msc_device.c:152
152       TU_ASSERT(p_request->bmRequestType_bit.type == TUSB_REQ_TYPE_CLASS);
(gdb) bt
#0  0x20091544 in mscd_control_request (rhport=0 '\000', p_request=0x1001ff4c) at ../../lib/tinyusb/src/class/msc/msc_device.c:152
#1  0x2008fcc0 in process_control_request (rhport=0 '\000', p_request=0x1001ff4c) at ../../lib/tinyusb/src/device/usbd.c:671
#2  0x2008f2b0 in tud_task () at ../../lib/tinyusb/src/device/usbd.c:386
#3  0x200957d8 in usb_background () at ../../supervisor/shared/usb/usb.c:80
#4  0x20096304 in run_background_tasks () at background.c:51
#5  0x2008a138 in run_code_py (safe_mode=NO_SAFE_MODE) at ../../main.c:268
#6  0x2008a6e0 in main () at ../../main.c:464
(gdb) p/x *p_request
$35 = {
  {
    bmRequestType_bit = {
      recipient = 0x2,
      type = 0x0,
      direction = 0x0
    },
    bmRequestType = 0x2
  },
  bRequest = 0x1,
  wValue = 0x0,
  wIndex = 0x83,
  wLength = 0x0
}
(gdb)
```

Is this expected behavior?

hathach, 2019-11-27:
@xobs it is a surely BUG. Previously, usbd only forward class specific or interface targeted request to class driveer. However, when adding support for USBTMC class, the class need to reset/re-init buffer when endpoint got stalled and clear stall, so It makes sense to also forward STD endpoint request to class driver as well, which falls into this TU_ASSERT() 

https://github.com/hathach/tinyusb/blob/master/src/device/usbd.c#L663

But I "forgot" to update the class driver that doesn't use the std endpoint request to silently drop it. I will push an fix soon enough.

hathach, 2019-11-27:
@xobs #230 should fix this, please pull and try again :)

closed 2019-11-27 by hathach

xobs, 2019-11-27:
I confirm this fixes the issue.  I can now emulate a failed disk and tinyusb doesn't panic.  Thanks for the fix!

hathach, 2019-11-27:
thanks for reporting, it is normally hard to get an stall from host. Let's me know if you need my help to troubleshoot why host stall the msc endpoints

xobs, 2019-11-27:
The stall happened because I hadn't yet implemented SPI flash in circuitpython, so it kept reporting the size as `0`.  The host thought the disk was going bad, so it tried to reset it after a few *minutes*.

It's fine now that I have storage implemented, but I wanted to make sure the issue wasn't caused by memory corruption.

hathach, 2019-11-27:
I am glad to hear that you are getting circuitpython ported to fomu :+1:

commit fe1eadf1 (2019-11-27): fix #229
commit 2e0dcd50 (2019-11-27): Merge pull request #230 from hathach/develop

fix #229