hathach/tinyusb #239: Add support for High Speed USB HID

PTS93, 2019-12-20:
Given that we are getting now a few devices that support High Speed USB I think it would be great to add support for that in the HID class.

High Speed devices can have a polling rate of 125us as opposed to 1ms with Full Speed.
As well as a much higher maximum payload size of 1024 vs 64.

This mostly relates to generic HID in/out I guess but should also affect devices that require very low latency interaction.
From what I see this shouldn't be a lot of work. Basically an additional descriptor and 1-2 functions to handle polling in us in addition to ms.

hathach, 2019-12-20:
@PTS93 i think the current stack support highspeed transferring with HID generic as well. Descriptors, endpoint size, report size are all user's configuration already. I don't know what needs to be added now :thinking:

PTS93, 2019-12-20:
Yea, I checked again and I misunderstood how the polling rate is set. Its defined as frames rather than a fixed time frame. So setting poll rate to 1 is 125us for a High Speed device and 1ms for a Full Speed device.
In that case we do not need any changes at all.

closed 2019-12-20 by PTS93