hathach/tinyusb #242: NXP submodule is not a git repository

hierophect, 2019-12-23:
**Describe the bug**
When checked out on 7a05b177 attempting to issue git submodule update results in the following error: 
```
fatal: not a git repository: /Users/lucian/Documents/Firmware_Projects/circuitpython/lib/tinyusb/hw/mcu/nxp/../../../../../.git/modules/lib/tinyusb/modules/hw/mcu/nxp
Failed to clone 'hw/mcu/nxp'. Retry scheduled
BUG: submodule considered for cloning, doesn't need cloning any more?
Submodule path 'hw/mcu/microchip/samd/asf4': checked out '82fe3aa05bc3b9f4956ee775c748a3885443f66b'
Submodule path 'hw/mcu/nordic/nrfx': checked out '7a4c9d946cf1801771fc180acdbf7b878f270093'
fatal: not a git repository: /Users/lucian/Documents/Firmware_Projects/circuitpython/lib/tinyusb/hw/mcu/nxp/../../../../../.git/modules/lib/tinyusb/modules/hw/mcu/nxp
Unable to fetch in submodule path 'hw/mcu/nxp'
```

The bug blocks all further requests for `git status` with 
`fatal: not a git repository: hw/mcu/nxp/../../../../../.git/modules/lib/tinyusb/modules/hw/mcu/nxp`
Until the tinyusb submodule is deleted.  

Linking @deshipu as he was having the same issue.

tannewt, 2019-12-23:
Try deleting the directory and then doing `git submodule update --init`. This problem should go away once all of our branches are updated.

hathach, 2019-12-24:
yeah, @hierophect I re-organize the hw/mcu/nxp submodules a bit to add support for iMX RT, I rename it from **lpc_driver** to `nxp_driver`. It probably confuse git & its cache. If simply remove the folder as Scott suggestion doesn't works, you may want to completely remove the previous submodule with following this guide https://gist.github.com/myusuf3/7f645819ded92bda6677

hierophect, 2019-12-30:
I did have to go in and delete the cache as well. Thanks for the help.

hierophect, 2020-01-17:
Unfortunately, I can't replicate the fix on my new machine. Even cloning from scratch, I'm getting this same error, and clearing the cache is not helping:

```
fatal: not a git repository: /Users/hierophect/Documents/Firmware_Projects/circuitpython/lib/tinyusb/hw/mcu/microchip/../../../../../.git/modules/lib/tinyusb/modules/hw/mcu/microchip
Failed to clone 'hw/mcu/microchip'. Retry scheduled
fatal: not a git repository: /Users/hierophect/Documents/Firmware_Projects/circuitpython/lib/tinyusb/hw/mcu/nxp/../../../../../.git/modules/lib/tinyusb/modules/hw/mcu/nxp
Failed to clone 'hw/mcu/nxp'. Retry scheduled
BUG: submodule considered for cloning, doesn't need cloning any more?
fatal: could not get a repository handle for submodule 'hw/mcu/microchip'
Failed to recurse into submodule path 'lib/tinyusb'
```

My current steps are:

- delete the tinyusb folder from lib/tinyusb
- run `git rm --cached lib/tinyusb`
- run `git submodule update --init --recursive` //will not work, since tinyusb is marked for deletion
- git reset --hard (attempt to remove unwanted changes to master, takes me back to the start)

My current issue is that the deletion of tinyusb seems to be automatically staged for commit, despite me never having done so, and thus the submodule update at the end will never download unbroken version? 

I apologize if this is just insufficient git fu and not project specific, but I have never encountered this issue before with another submodule and I don't see many related results on Google. Is there something in particular that's making these folders particularly prone to messing up git's cache?

closed 2020-01-17 by hierophect