hathach/tinyusb #247: samd SETUP buffer size

kaysievers, 2019-12-26:
I need to add something like this:

```diff
--- a/tinyusb/src/portable/microchip/samd/dcd_samd.c
+++ b/tinyusb/src/portable/microchip/samd/dcd_samd.c
@@ -44,6 +44,8 @@ static inline void prepare_setup(void)
   // Only make sure the EP0 OUT buffer is ready
   // SETUP token doesn't check any other parameters
   sram_registers[0][0].ADDR.reg = (uint32_t) _setup_packet;
+  sram_registers[0][0].PCKSIZE.bit.MULTI_PACKET_SIZE = sizeof(_setup_packet);^M
+  sram_registers[0][0].PCKSIZE.bit.BYTE_COUNT = 0;^M
 }
```

 to make https://github.com/hathach/tinyusb/commit/49b316143ad4e1bd6130b3032b8418be7e0bc2c3 work properly on Linux.

hathach, 2019-12-27:
fixed, thanks for the issue. hopefully this is the last PR for racing issue.

closed 2019-12-27 by hathach

kaysievers, 2019-12-27:
Yeah, updated all the repos now and tested a SAMD51 device connected to Windows, Android, Linux (Desktop), iOS. It seems all fine. Thanks!

commit 89f3ad59 (2019-12-27): fix #247
commit dda4c9a9 (2019-12-27): Merge pull request #248 from hathach/develop

fix #247